riscv/riscv-cheri #4: Splitting `Zcheri_purecap` into 2 different extensions: `Zcheri_purecap` and `Zcheri_protection_rings`

vmurali, 2024-01-22:
Thanks for submitting the specification draft for CHERI!

CHERI allows us to build secure systems where a protection ring (M, H, S, U modes) is subsumed using just capability bounds (for accessing privileged memory) and ASR permissions (for accessing privileged registers).

In fact, ASR permissions are better than M-mode privileges. For instance consider a (privileged) trap handler calling a non-privileged function subroutine. If the trap handler wants to call the unprivileged function, it has to set up the privilege stack so that it can `MRET` into the unprivileged subroutine while lowering the privilege, and the subroutine returns with an `ECall` that raises the privilege when jumping back to the trap handle; and one has to maintain some state in the trap handler to direct the trap handler to go back to the instruction after the unprivileged function call in the trap handler.

With ASR permissions, the trap handler with ASR permissions will call the subroutine without ASR permission and after returning, the trap handler gets back the ASR permissions through the linked register which captures the original ASR permissions of the trap handler.

Regarding the CSRs (enabled with Zicsr extension), there are a few CSRs that have to be present even in a CHERI system without a protection ring - namely

1. `MTVECC`
2. `MEPCC`
3. `MSratchC`
4. `MCause`
5. `MTVal`

These registers are necessary to handle at least the exceptions introduced by CHERI. So these
should also be part of `Zcheri_purecap`, along with the `MRet` instruction that allows the trap handler to return to the next-scheduled hart.

Based on the above, all mentions of M-mode or S-mode should be removed from Chapter 3 (that describes `Zcheri_purecap`) and put into a new chapter (with a new extension for protection rings in CHERI, aka `Zcheri_protection_rings`, which contains the rest of the chapter, including supervisor-mode CSRs, etc, and other registers that are not essential like `MISA`).

tariqkurd-repo, 2024-01-22:
We're not considering any refactoring of extensions before we have had feedback from the RISC-V Architecture Review Committee

vmurali, 2024-01-22:
Okay, so what is the procedure to provide feedback or request for changes to this spec?

vmurali, 2024-01-23:
> I wish I could comment more substantively about this proposal; is the assumption that someone would want to enable a purecap CHERI-RISC-V with no exception support at all?

The processor we are interested is a purecaps, no protection rings, single address space shared by all processes. We *do need* exception support (and hence the few CSRs (`MScratchC`, `MEPCC`, `MTVecC`, `MCause` and `MTVal`) to deal with exceptions, and the instructions `MRET` and hopefully `ECALL`/`WFI` to voluntarily yield). But we don't need privilege mode to handle the exceptions - ASR permissions (and capability memory bounds) serve that role.

NB:
1. With such a processor, one could also evaluate how much power/performance/area is improved while offering much better security with capabilities (compared to traditional MMUs/PMPs/PMAs). Such a purecaps system would provide a true apples-to-apples comparison with traditional systems.

2. A single address space shared between all processes is sufficient (even for systems that need disk swap space that are traditionally implemented using MMUs) because of the ideas in Issue https://github.com/riscv/riscv-cheri/issues/5

sorear, 2024-01-24:
We already have this - support for privilege modes other than M is optional in the base privileged ISA, so if you want to implement a chip with only M mode, only interrupt handling registers for a single mode, and relying on ASR exclusively for sandboxing, you can.

vmurali, 2024-01-24:
Right. But the way Chapter 3 is organized, it appears as if all the CSRs (and S mode) are necessary to be `Zcheri_purecap` compatible. It would be good to specify what exactly is necessary (which is only the 5 CSRs that I mention) and the `MRet` instruction (and optionally `ECall/WFI`). If there are other necessary registers, we can add those.

andresag01, 2024-01-25:
@vmurali : Thanks for the review comment. The intention of the current specification is not generally for CHERI to require implementing features that are optional in the privileged RISC-V ISA. I agree that we need to make this clearer in the text by, for example, indicating that a CSR is only extended to CLEN if it exists/is needed in the first place according to regular RISC-V rules. However, I do not expect that this requires changing the split of the CHERI extensions.

jrtc27, 2024-01-25:
> > However, I do not expect that this requires changing the split of the CHERI extensions. Wouldn't it be sufficient to simply add clarifications to the current spec as needed?
> 
> I believe RISCV base unprivileged spec doesn't have to handle exceptions and only if you implement portions of the privilege spec (virtual memory, pma, PMP, S mode etc) you need to start handling exceptions.

Yes it does. From ECALL and EBREAK, for example:

> These two instructions cause a precise requested trap to the supporting execution environment.

The base spec requires that certain things trap. What trapping means, including how the cause is reported, is left to the privileged spec to define.

tariqkurd-repo, 2025-06-11:
@vmurali is happy for me to close this

closed 2025-06-11 by tariqkurd-repo