riscv/riscv-cheri #5: Rename `Zcheri_pte` to `Zcheri_pte_traditional`, and introduce `Zcheri_pte_single_address_space`

vmurali, 2024-01-22:
The traditional virtual memory mapping serves the following purposes:

1. Isolation between processes by giving each process a separate virtual address space (0 to 2^XLEN-1 virtual addresses for each process).
2. Protection within a process using Read/Write/Execute permissions.
3. A policy to replace parts of the physical memory with some other data/code (tracked using "accessed" and "dirty" bits)

CHERI can provide process isolation using capability bounds on a single address space shared between all processes. Protection within the process is provided using CHERI permissions. So we could have an alternative virtual memory mapping where a PTE entry contains just "valid", "accessed" and "dirty" bits, and all processes share a single address space. This could be a new extension `Zcheri_pte_single_address_space`.

In terms of hardware, this obviates the need for virtual memory translation for even L1 cache accesses, because all caches can be virtually tagged. In traditional virtual memory, virtual tagging of caches is not possible because of aliasing in virtual memory and hence the translation is done for even L1 cache accesses. In the new scheme provided by Zcheri_pte_single_address_space, virtual memory translation has to be done only on last level cache miss (LLC miss).

(Allotting a new physical page for a virtual page should also check if that virtual page has already been mapped to avoid aliasing. This can be achieved using a reverse mapping, from physical memory to a single address space virtual memory).

tariqkurd-repo, 2024-01-22:
We're not considering any refactoring of extensions before we have had feedback from the RISC-V Architecture Review Committee

vmurali, 2024-01-22:
Okay, so what is the procedure to provide feedback or request for changes to this spec?

jonwoodruff, 2024-01-23:
@vmurali This is certainly a cool idea, and would by synergistic with some OS ideas out there (e.g. CheriOS ([[PDF] cam.ac.uk](https://www.repository.cam.ac.uk/bitstream/handle/1810/326715/le277thesis_final.pdf))).

This spec itself is a core set of CHERI features that are ready to be ratified/commercialised.  We expect quite a few extensions, many research-only, some commercialised, and some to be ratified in further extensions.  There is a lot going on!  But I guess this spec itself isn't the place for fresh ideas.  There is a CHERI slack channel where people are socialising new CHERI ideas, and that might be a good place for this sort of discussion:

https://join.slack.com/t/cheri-cpu/shared_invite/zt-2azp67ap2-gqbKWzuqtLYm_IWCLGdYHA

vmurali, 2024-01-23:
I agree this specific extension proposal is orthogonal to the current version of the spec. Perhaps this issue can be closed and I can join the slack channel.

sorear, 2024-01-25:
This is orthogonal to CHERI; RISC-V supports the use of virtual caches through the ASID mechanism, there's been some talk of improving ASIDs to allow multiple cores to share a virtual cache but I haven't heard anything recently and certainly not here.

vmurali, 2024-01-25:
With traditional virtual memory, one cannot guarantee non-aliasing into shared libraries/resources (unless you are talking about a single-address space, but that voids the memory protection given by separate virtual address spaces using traditional MMUs).

CHERI provides memory protection while allowing a single-address space shared between all processes. So CHERI basically enables such an architecture.

But I agree that single-address space virtual memory doesn't contradict with whatever is written in the spec; it can be in a custom extension.

jrtc27, 2024-01-25:
You missed:
4. Virtualise the address space so you don't have to deal with holes, high addresses, etc and can instead choose the exact layout of your address space
5. Be able to swap pages in and out without having to make all your applications aware of it and support pointers changing address (which breaks C, or at least real-world C; e.g. you can no longer use addresses as keys in a hash table)
6. Support copy-on-write semantics, page sharing, etc.

Even in a single-address-space OS, you probably want at least some non-empty subset of these, and all require full PTEs.

vmurali, 2024-01-25:
> You missed:

> 4. Virtualise the address space so you don't have to deal with holes, high addresses, etc and can instead choose the exact layout of your address space

> 5. Be able to swap pages in and out without having to make all your applications aware of it and support pointers changing address (which breaks C, or at least real-world C; e.g. you can no longer use addresses as keys in a hash table)

> 6. Support copy-on-write semantics, page sharing, etc.
> 
> Even in a single-address-space OS, you probably want at least some non-empty subset of these, and all require full PTEs.

I agree that traditional virtual memory combines the implementation for 1 and 2 with 4, 5 and 6. That's exactly why I am proposing a single-address space virtual memory, so that we get 4, 5 and 6 without 1 and 2. 1 and 2 are provided by CHERI.

vmurali, 2024-01-25:
(6) Page sharing is not something we would need to support in a single-address space virtual memory.

jrtc27, 2024-01-25:
> I agree that traditional virtual memory combines the implementation for 1 and 2 with 4, 5 and 6. That's exactly why I am proposing a single-address space virtual memory, so that we get 4, 5 and 6 [...]

So what are you throwing out that doesn't break 4 and 5? Those need a PA in your PTE, at which point who cares about a few extra permissions.

vmurali, 2024-01-25:
> Those need a PA in your PTE, at which point who cares about a few extra permissions.

Yes, I need a PA in my PTE. I am not trying to save on the few bits. Having these bits may lead to contradictions with CHERI's permissions and in general extra unnecessary checking (in software or hardware).

sorear, 2024-01-25:
Introducing a "hardware trusts software to not create virtual page aliases, else nasal demons" rule seems at odds with CHERI's design principles of hardware-assisted checking and minimizing the trusted software base.  If you do build hardware alias checking, it's not complicated to extend that to full alias support (when you bring in a cache line, evict the aliases) and then you can use normal SvNN.

> https://join.slack.com/t/cheri-cpu/shared_invite/zt-2azp67ap2-gqbKWzuqtLYm_IWCLGdYHA

I just joined and I don't see any discussion of Zcheri, or architectural design in general, in the past few months, is that the right forum?

vmurali, 2024-01-25:
> Introducing a "hardware trusts software to not create virtual page aliases, else nasal demons" rule seems at odds with CHERI's design principles of hardware-assisted checking and minimizing the trusted software base.

I agree that it increases CHERI's TCB. The virtual memory mapper (I am envisioning in SW, but it can also be in HW) must gain trust somehow, for instance, be formally proven to be correct (including avoiding aliases).

The problem with traditional virtual memory is that aliases cannot be avoided - we can only detect them and take evasive actions (like evicting the aliases). Whereas, with single-address space, aliases can be avoided with a simple `Physical Page -> Bool` mapping (with trust in the SW that handles the mapping).

jonwoodruff, 2024-01-25:
> Introducing a "hardware trusts software to not create virtual page aliases, else nasal demons" rule seems at odds with CHERI's design principles of hardware-assisted checking and minimizing the [...]

@vmurali This is a slack with lots of us CHERI people. It started with Morello support, but also has a cheri-riscv-hardware channel.  If you would like to discuss "single-address space systems" or similar, maybe request a new channel.  Could maybe start with some questions in "general" and if lots of discussion is generated, request a new channel and pull the interested parties with you.  Discussions more directly about this spec are fine here, but there should be plenty of appetite to discuss newer ideas/extensions with CHERI people over there.

vmurali, 2024-01-25:
Thanks @jonwoodruff I will get around to posting there.

andresag01, 2024-09-27:
@vmurali : This ticket has not been updated for a while and it looks like the discussion moved to Slack. Can we close it?

tariqkurd-repo, 2025-06-11:
@vmurali is happy for me to close this

closed 2025-06-11 by tariqkurd-repo