riscv/riscv-cheri #9: Order of checks for CJALR & invalid address handling not specified

Timmmm, 2024-01-22:
The spec for CJALR says

> The target capability may have Invalid address conversion performed and is then installed in pcc.

but it doesn't say whether invalid address conversion occurs before or after bounds checking (or the other checks, but those are unlikely to be affected by invalid address conversion).

IMO it would be more elegant to do it afterwards - it means most of the CJALR implementation doesn't need to care about invalid addresses; it's all neatly contained in the bit at the end where you install the cap in PCC. Also I imagine it would simplify implementations because there's no data dependency - you can do the bounds check without having to wait for invalid address conversion.

The security properties are the same, since invalid address conversion clears the tag if necessary. The only observable difference would be that in some cases you would get a tag violation on the next cycle, instead of a length violation on the current cycle. I think that's ok though.

Timmmm, 2024-01-22:
@tariqkurd-repo has clarified that the intention is to do bounds checking after converting the address. The benefit is that you don't need to do the whole invalid address conversion procedure:

* You don't need to clear the tag if the cap is sealed, since CJALR always unseals the cap.
* You don't need to do a representability check since if it is in bounds it must be representable.

I think this should still be clarified in the spec though.

tariqkurd-repo, 2024-01-22:
Feel free to make a PR with proposed wording improvements @Timmmm

PR #10 (2024-01-31): Clarify CJALR operation order
Clarify that length violations are checked on the address after it has undergone invalid address conversion.

Also fix a typo in the title.

Fixes #9
commit 9c862a6b (2024-01-31): Clarify CJALR operation order

Clarify that length violations are checked on the address after it has undergone invalid address conversion.

Also fix a typo in the title.

Fixes #9
closed 2024-01-31 by andresag01