riscv/riscv-cheri #15: Extension names are all invalid

jrtc27, 2024-01-23:
All extension names in the spec are Zcheri_foo, but extensions can't have _ in them, because it's the separator character.

tariqkurd-repo, 2024-01-23:
Hi Jessica - glad you're back. All the extension names are also invalid because Zc* relates to 16-bit encodings.
We assigned temporary names so that the intent was clear and will change them when the ARC advise correct ones.
There's no point in trying to come up with "official" ones as they will be changed - this is my Zc experience.

jrtc27, 2024-01-23:
Sure, the semantic meaning suffers from that, but it's probably best to not pick something as a placeholder that's so blatantly invalid? Or at least add a note if that's intentional?

andresag01, 2024-01-23:
Hi @jrtc27, you are correct: the names are invalid. Not a lot of thought has gone into this because, as @tariqkurd-repo mentioned, it would be good to have ARC advise. We expect the names to change, so most instances of that in the adoc are actually variables that are very easy to change (see [here](https://github.com/riscv/riscv-cheri/blob/main/src/riscv-cheri.adoc?plain=1#L44)).

tariqkurd-repo, 2024-01-23:
That text was in there but looks like it didn't make the final version.

arichardson, 2024-01-24:
> Hi @jrtc27, you are correct: the names are invalid. Not a lot of thought has gone into this because, as @tariqkurd-repo mentioned, it would be good to have ARC advise. We expect the names to change, [...]

I agree that using invalid names is not great even if they are placeholders. Maybe just drop the _ for now?

Using the variable is great and will make it easier to change, but I noticed that the usage in headings will still need to be modified since asciidoc does not appear to substitute it there.

closed 2024-04-20 by arichardson