riscv/riscv-cheri #17: Zcheri_pte lacks support for load-side revocation

jrtc27, 2024-01-23:
The current state of the art revocation scheme, and the only one present in CheriBSD releases, uses a load-side, which requires the ability to conditionally trap on loads based on a 1-bit generation counter, not present in the current draft.

tariqkurd-repo, 2024-01-23:
Hi Jessica - Zcheri_ptr is current WIP - and please extend it.

andresag01, 2024-01-23:
Agreed, the PTE extension is provisional and very much a work in progress. The current spec highlights this in a note and that the current proposal is missing features that exist in CHERI v9.

jonwoodruff, 2024-01-23:
Not to muddy the waters, but one could view the load-side generation barrier as an independent feature that could be useful without CHERI for other types of garbage collection.  Is there a possibility of this?  Is there a better chance of the PTE bit being allocated this way if it had support from a wider community?  (Presumably there is a [garbage collection SIG](https://github.com/riscvarchive/groups/blob/main/Managed-Runtimes/CHARTER.md)?)

PeterRugg, 2024-01-23:
Just noticed this as well. I also wonder about `cloadtags`. We really need to get stats on how much that matters with the load-side barrier, but in Cornucopia it looks like it saved about 50% for xalanc and 20% for omnet, which are the only intensive benchmarks, and I doubt there's too much change there between store-side and load-side in terms of `cloadtags` saving since the core intra-page sweeping loop is the same.

closed 2024-09-16 by andresag01