riscv/riscv-cheri #18: Cbuildcap and ctestsubset and the use of DDC

francislaus, 2024-01-23:
Both cbuilcap and ctestsubset use DDC as a parameter if the cs1 == 0. However, both instructions only have Zcheri_purecap as their prerequisite. DDC is a register that is only defined in Zcheri_legacy, if I understand the spec correctly. Thus, there exists a case where the specified behaviour of the instruction is not possible.

These are the two straightforward solutions that came into my mind.

1. Cbuildcap and ctestsubset use c0 if cs1 == 0
2. DDC is defined in Zcheri_purecap

There probably are other solutions. Given the two I named, the first one seems way more clean. This would not optimally use the encoding space of the register index for cs1, but it would prevent DDC from being defined in purecap where it is not needed.

andresag01, 2024-01-23:
@francislaus : Thanks for the feedback. You are absolutely right, the original specification for CBuildCap and CTestSubset in CHERI v9 does have a special case for DDC when `cs1 == 0`. This was not taken into account in the current definition of CBuildCap and CTestSubset and needs to be fixed.

Given the current split of extensions in the spec, option (1) seems is a better fit.

tariqkurd-repo, 2024-01-23:
I also vote for option (1) - as for a purecap core we don't want to include DDC at all.
Can you file a PR @francislaus with the fix? thanks!

francislaus, 2024-01-23:
@tariqkurd-repo @andresag01 Sounds good. I will file a PR this afternoon.

arichardson, 2024-01-24:
I think using cnull instead of DDC should be fine. The use of DDC is a carryover from the MIPS spec where we also had that behaviour for other instructions (loads/stores) to save encoding space. For consistency, c0 probably makes sense. Maybe those encodings should be marked as HINTs?

jrtc27, 2024-01-24:
They're not hints though, they write a value to the destination register (with the exception of cd=c0, which is true regardless of what source operands you have).

arichardson, 2024-01-24:
> They're not hints though, they write a value to the destination register (with the exception of cd=c0, which is true regardless of what source operands you have).

Use of CNULL doesn't really make much sense though so maybe we could just make those encodings RESERVED?

francislaus, 2024-01-25:
I could get behind the RESERVED encodings. This is done over an over in RISC-V where it does not make sense to have x0 as an operand.

andresag01, 2024-02-01:
Apologies for closing this ticket. GitHub automatically closed it because it unhelpfully.

PR #82 (2024-02-07): Cbuildcap and ctestsubset do not use DDC
This should close #18 

I decided to not make cs1==0 RESERVED because it is a valid input.
closed 2024-02-07 by andresag01