riscv/riscv-cheri #25: Permission encoding order mismatch

Timmmm, 2024-01-23:
For RV64 the ADP bits are encoded like this:

[image]

However `CGetPerms` and `CAndPerms` expect this order:

[image]

The table for encoding RV32 permissions uses the same order (reversed):

[image]

It would be much better if these were consistent!

Also did this question get resolved?

[image]

andresag01, 2024-01-24:
@Timmmm : Thanks for pointing this out; indeed, it needs to be fixed! Would you like to create an MR for this?

The `TODO` has not been resolved.

Timmmm, 2024-01-25:
@tariqkurd-repo you mean remove the TODO and leave as-is? If new useful combinations are found what's the plan for RV32? Use the reserved pattern?

PR #36 (2024-01-25): Make permission bit order consistent
Use the same ordering for permission bits in the RV64 encoding as in CAndPerms etc.

Fixes #25
commit 69d93532 (2024-01-25): Make permission bit order consistent

Use the same ordering for permission bits in CAndPerms as in the RV64 encoding.

Fixes #25
closed 2024-01-25 by andresag01