riscv/riscv-cheri #31: CSETMODE is probably dispensible

sorear, 2024-01-24:
Unlike CANDPERM or CRAM which are critical for memory allocation, CSETMODE only seems to be used when setting up pointers to functions, which typically only happens during program loading or JITing.  For an operation that's unlikely to be done more than 1e5 times in a program execution, we can probably lower it to:

    CGETHIGH temp0, in
    BINVI temp0, temp0, 25|47
    CSETHIGH temp0, in, temp0
    CBUILDCAP out, in, temp0

tariqkurd-repo, 2024-01-24:
Yes - but there is a big performance difference between the two sequences.
A minimal ISA set without `csetmode` would be acceptable.

sorear, 2024-01-24:
The question is whether the dynamic counts of CSETMODE in real-world applications are high enough for anyone to notice replacing them with a four-instruction sequence.  I'm not proposing to define a new extension, either CSETMODE is a frequently used instruction or it isn't.

tariqkurd-repo, 2024-01-24:
there are many such examples - analysis of the frequency of usage of all instructions is planned

jonwoodruff, 2024-01-25:
In dynamic traces of test workloads of specint benchmarks, pure-cap, statically linked, CSetFlags (the old equivalent of SetMode) did not appear even once.  I imagine this implies that we're only successfully tracing userspace (I'm relying on the objdump of the binary to lookup the actual instructions, so kernel instructions won't be found); CSetFlags was surely used by the loader at some point, for example.

A static count of a pure-capability kernel found 10 instances of CSetFlags.

jonwoodruff, 2024-01-25:
Also, for something like the loader, one could keep an executable root capability for each mode, and just choose the right one for deriving executable capabilities, avoiding even the very rare uses of this instruction.

jrtc27, 2024-01-25:
CSetMode is definitely used by rtld, you must not be tracing correctly

jonwoodruff, 2024-01-25:
@jrtc27 These were statically linked benchmarks; I presume tracing began after the benchmark began running and after loading.  (These were traced by Mingle) Is it conceivable that rtld did not run after the benchmark began executing?

jrtc27, 2024-01-25:
Static linking by definition means no rtld

andresag01, 2024-09-27:
The latest version of CHERI RISC-V allows using most of the ISA extension with a different encoding. However, this requires that we provide instructions to perform all capability manipulation operations without having to manually fiddle with the metadata and then using CBLD, so the CSETMODE (now SCMODE) must remain and we added related instructions like GCMODE, GCTYPE, etc even though they are probably used infrequently.

closed 2024-09-27 by andresag01