riscv/riscv-cheri #34: M bit sense should be inverted

sorear, 2024-01-24:
In Zcheri_purecap, M is a reserved bit, so Zcheri_purecap software will set M to 0.  For this reason, if Zcheri_mode is enabled, M = 0 should trigger Zcheri_purecap-compatible behavior, with M = 1 used for the non-Zcheri_purecap-compatible behavior.

sorear, 2024-01-24:
I am not proposing for purecap software to rely on the state of the bit, or look at the bit at all.

Purecap software sets all reserved bits to 0, so extensions should do something with the value "0" that is appropriate for purecap.  Software that uses a given extension will set its corresponding allocated bits to "1".

jrtc27, 2024-01-24:
> So software for a purecap only core must not be relying on the state of that bit.

That doesn't work; if the bit exists it needs to be set to the right value! Which makes 0 the sensible value to use in this world, so you don't have to do anything to get the right value by default. Only if you want legacy mode do you need to set it to 1.

sorear, 2024-01-28:
My thoughts on the matter have shifted to "capability extensions can redefine the format of the metadata word within fairly broad limits, so although it is architectural software should treat the metadata word format as implementation defined unless the exact ISA string is known; cbuildcap should not be used in generic software on anything other than a bit pattern previously generated by the implementation".  If that is adopted, this issue can be closed.

andresag01, 2024-09-27:
The sense of the M-bit has already been inverted: https://github.com/riscv/riscv-cheri/pull/305

closed 2024-09-27 by andresag01