riscv/riscv-cheri #37: Behaviour of CRAM not specified for malformed caps

Timmmm, 2024-01-24:
It's not clear to me what CRAM should do when operating on a malformed cap (i.e. E < 0). Please don't say it's implementation defined!

From a Sail perspective, treating E<0 as E=0 would be easiest. I believe the existing code treats it as CAP_MAX_E:

```
function getRepresentableAlignmentMask(len : xlenbits) -> xlenbits = {
  let (exact, c) = setCapBounds(default_cap, to_bits(sizeof(xlen), 0), 0b0 @ len);
  let e = min(unsigned(c.E), cap_max_E);
  let e' = if c.internal_E then e + internal_E_take_bits else 0;
  ones(sizeof(xlen)-e') @ zeros(e')
}
```

I was planning on changing it to this (note the `c.E` type is changed to an `int` instead of `bits` so it can actually be negative here; the previous implementation wrapped):

```
function getRepresentableAlignmentMask(len : xlenbits) -> xlenbits = {
  let (_exact, c) = setCapBounds(default_cap, to_bits(sizeof(xlen), 0), 0b0 @ len);
  if c.internal_exponent then {
    let e = max(c.E, 0);
    ~(zero_extend(ones(e + cap_xe_width)))
  } else zeros()
}
```

Not tested.

sorear, 2024-01-24:
Given that CRAM takes an integer input, and not a capability, I think it's well defined regardless of the details of the capability encoding.

Timmmm, 2024-01-24:
Ah good point - E can never be negative here since `setCapBounds` won't generate malformed caps. I'll change it to an assert.

closed 2024-01-24 by Timmmm