riscv/riscv-cheri #44: Debug GPR access

sorear, 2024-01-25:
The debug spec allows GPRs to be accessed through the Access Register abstract command, and non-GPRs either through abstract commands or by running program buffer fragments.  Access Register can use an access size of 32, 64, or 128, but 65 and 129 are not options; how does a debugger read or write capabilities in GPRs, or restore capabilities in GPRs after using the GPR for a program buffer fragment?

System bus access also needs to be extended for CLEN-bit access, but it is an optional feature.

tariqkurd-repo, 2024-01-25:
The intention is that program buffer code is used to read the integer value, then `cgethigh` then `cgettag` to read the fields of the CLEN registers, so that the debug interface doesn't change.

System bus as specified will work without changing the width, but may not be able to monitor tags.

sorear, 2024-01-25:
That would work if a split capability register file were used, but with our merged file, you cannot do a `cgethigh` or `cgettag` without overwriting a register and you can't overwrite a register until you've read both its metadata and its tag.

tariqkurd-repo, 2024-01-25:
yes - so we want to stuck with the standard debug specification - so I believe the documented scheme works correctly.

sorear, 2024-01-25:
dscratchN is described as "The debugger can make no assumptions about the contents of these registers between commands", so if the debugger cannot save a GPR in dscratchN between abstract commands.  It might be possible to use a single abstract command to completely deconstruct a GPR into dataN registers, but this would definitely go over the 2 instruction mandatory minimum size of the program buffer.  Moreover, both dscratch registers are explicitly defined as optional both here and in the debug spec.

jrtc27, 2024-01-25:
There needs to be an extension to abstract commands (in fact our cores _only_ support that), but it's not hard to envisage how that might look.

sorear, 2024-01-25:
I was briefly considering proposing a "DGETHIGH" instruction that deconstructs a capability while preserving the tag, and can also be used to construct Infinity capabilities while avoiding the can of worms that is pc/ddc access in debug mode, but this doesn't help with the reverse problem of building a capability in a single GPR while not touching the other 30 GPRs.

I would rather avoid mandating dscratch[01]c and sufficient program buffer and data register sizes to send a program using csrr/csethigh/cbuildcap, as that would involve hundreds of bits of new state.

Given that the Access Register abstract command is required for GPRs anyway, I think the most direct route is to allocate 64 abstract register numbers for "HIGH(xN)" and "TAG(xN)"?

sorear, 2024-02-02:
I was overthinking this, it can be handled using standard-defined mechanisms and a non-outrageous amount of resources.  Note that Flute and Toooba do **not** satisfy the requirements documented in #88, since they rely on dscratch0 being preserved between program buffer executions and there isn't a way for the debug module to express that guarantee to debuggers yet.

andresag01, 2024-04-15:
@sorear : Thank you for your contribution. Your PR has been merged clarifying the debug requirements for CHERI, so this ticket can be closed.

closed 2024-04-15 by andresag01