riscv/riscv-cheri #51: constructing capabilities in debug mode

sorear, 2024-01-25:
This specification says that non-monotonic capability operations are performed in debug mode by using pcc as a source of Infinity.  However, all instructions that read pcc are illegal in debug mode (this is an intentional feature of the debug spec, as it is implementation defined whether the program buffer physically exists in memory or feeds instructions into the core through another mechanism; in the latter case, it is assumed that pc does not update and "dpc" accesses the physical pc while it is stopped).

andresag01, 2024-01-25:
> However, all instructions that read pcc are illegal in debug mode

It is for this reason that the PCC is also accessible by reading a CSR address, which can be done from the program buffer. Also, when Zcheri_mode is supported, DDC is also a source of the Infinity capability.

andresag01, 2024-02-21:
Fixed by #113

closed 2024-02-21 by andresag01