riscv/riscv-cheri #55: Use named citations

sorear, 2024-01-26:
(RISC-V, 2022) or (RISC-V, 2023) does not clearly communicate which of the four cited RISC-V specifications is being used.  Can we use a different format that clearly and memorably distinguishes them?

tariqkurd-repo, 2024-01-26:
agreed - do you know how this works @andresag01 ?

andresag01, 2024-01-26:
Agree in principle. However, we are using this citation style because the original RISC-V document template is configured like this (see https://github.com/riscv/docs-spec-template). Are we permitted to change it?

sorear, 2024-01-26:
We may be running into early-adopter issues as the first asciidoc native spec to need to cite several other specifications (and I just added one more, Zicclsm is in riscv-profiles).  Perhaps someone could file an issue there to get official guidance.  I would do it but I don't know enough about asciidoc or bibtex to properly explain the issue.

jrtc27, 2024-01-26:
Inline citations tend to either be some combination of author+year or just an index. Once you start including the title it becomes unwieldy. Given the field I would normally default to the IEEE style, which is just a number; less informative but also less confusing?

jrtc27, 2024-01-26:
FYI https://github.com/riscv/riscv-cheri/blob/b87760e740dfaaaa783609a9d4960393a84eaec4/src/riscv-cheri.adoc?plain=1#L22 is where this comes from.

arichardson, 2024-01-26:
I would also prefer IEEE style, but the base priv and unpriv specs both use `:bibtex-style: apa`, so I assume we are required to match their style.

sorear, 2024-01-27:
We're writing a technical specification, not a research paper, the problem space is different.  [IETF](https://www.rfc-editor.org/rfc/rfc7322.html#section-4.8.6), [ISO](https://www.iso.org/ISO-house-style.html), and [W3C](https://www.w3.org/Guide/manual-of-style/#References) (the first three I checked) all recommend using inline citations that are not simply bibliography indices or author and year.

jrtc27, 2024-01-27:
Isn't the problem just that it's using the anti-pattern of "in <citation>" rather than "in <name of document> <citation>" (i.e. "in the RISC-V Zfoo extension [42]")?

sorear, 2024-01-27:
Perhaps, although ideally we'd standardize the phrase that we use to refer to Zfoo and then we might as well use that as a reference tag.

tariqkurd-repo, 2024-09-30:
I also don't really like the citations, but as the objective is to get merged into the RISC-V priv/unpriv specifications then we need to follow their scheme, and if that scheme is has problems then I'd advise filing an issue on the riscv-isa-spec

closed 2024-09-30 by tariqkurd-repo