riscv/riscv-cheri #57: M/S-mode ambient authority other than "system registers"

sorear, 2024-01-26:
Things you can do in S-mode or M-mode that you can't do in U-mode, _other_ than access memory or mwhatever/swhatever registers:

* Cause an "Environment call from [MS]-mode" exception using the `ecall` instruction.
* Wait for interrupts.  (Not really authority but treated like it)
* Execute `cbo.zero`, if `senvcfg.CBZE`=0.
* Execute `cbo.flush` or `cbo.clean`, if `senvcfg.CBCFE`=0
* Execute `cbo.inval`, if `senvcfg.CFIE`!=11
* Access the ("unprivileged") `seed` CSR, if `mseccfg.USEED`=0
* Access unprivileged `cycle`, `time`, `instret`, `hpmcounterN`, if the corresponding `scounteren` bit is 0

My feeling is that all of these should execute as if prv=U if pcc.ASR=0.  Thoughts?

jrtc27, 2024-01-26:
Well, the question is whether it’s about accessing state, per the name, or system control as a whole (note that Arm take it to be the latter in Morello and just call it System permission, which seems a sensible direction). Some of these seem sensible under the latter definition. Some of these would break the ability to run userspace without ASR, which CheriBSD does today; ecall in particular comes to mind, and that one can easily be enforced by software-defined policy in the handler, you don’t need hardware for it.

sorear, 2024-01-26:
Executing ecall in U-mode without pcc.ASR "as if [the current privilege were U-mode]" is a noop and shouldn't break the ability to run things.