riscv/riscv-cheri #58: pcc metadata after cm.j(al)t

sorear, 2024-01-26:
The jump table contains addresses, which raises the question of how the new pcc capability is derived from the address in the jump table.  I think the most useful option is to take the bounds from jvtc, so that it can be used as a subroutine library shared between many code capabilities.

tariqkurd-repo, 2024-01-26:
`jvtc` is intended to cover the bounds of the table to prevent access to it, which doesn't restrict its usage to that, but it doesn't seem appropriate to use it in this way.

The intention of the extension is to reduce code size, and so doubling the size of the table by making every entry a cap greatly reduces the benefit, this is something which has been mentioned before.

It would be possible to have another cap which provides the bounds, maybe with a bit in the LSB of each table entry to say whether to take PCC bounds or bounds of this other register.

sorear, 2024-01-26:
How did you get from "The jump table contains addresses" and "take the bounds from jvtc" to "the opener wants to redefine the jump table to make every entry a cap and take the bounds from capabilities stored in the jump table"?

andresag01, 2024-01-26:
The _jump via table_ instructions calculate the target address like this: `jvtc.address+index*XLEN/8`. So the question is where the new PCC's metadata is coming from -- this is not clear to me looking at the current spec. There are 2 options as far as I can see:

1. The bounds come from PCC, so the "new" PCC i.e. the target capability of the jump is (excluding legalization, etc):
```
newPCC = setCapAddr(PCC, jvtc.address+index*XLEN/8)
```
2. The bounds come from JVTC, so the "new" PCC is:
```
newPCC = setCapAddr(JVTC, jvtc.address+index*XLEN/8)
```
I think @sorear is advocating for (2). Right?

IMHO there are arguments for either of the options above. @tariqkurd-repo: In the existing Zcmt trials, how does the toolchain deal with the table? Where is it normally placed relative to the code? Of course, this will not consider CHERI, but at least is a bit more info.

@tariqkurd-repo: A other thoughts about Section 8.11 in the CHERI spec (i.e. Zcmt):
* It would be good to indicate that jvmtc needs to be Almighty if someone wants to make it look like they are running regular RISC-V. We have similar notes for PCC and DDC elsewhere
* Why must JVTC have X-perm? Are you assuming (2) above as well? In the case of (1) above, I would have expected JVTC to have only R-perm

sorear, 2024-01-26:
> I think @sorear is advocating for (2). Right? (I am assuming that by bounds you mean the whole cap metadata, and I am using both words interchangeably)

Correct on both counts.

> In the existing Zcmt trials, how does the toolchain deal with the table? Where is it normally placed relative to the code?

The table is created in a .riscv.jvt section with SHF_EXECINSTR permission, see riscv-non-isa/riscv-elf-psabi-doc#349 and llvm/llvm-project#77884.  Its placement is controlled by a linker script.  It is unclear to me how lld determines the section order in the absence of a linker script, and can only speculate how this will be used in OS environments.

> Why must JVTC have X-perm?

There are two intended implementation strategies for Zcmt in the base ISA.  One, cm.jt can perform an XLEN-bit load in the execution pipeline to the address (jvt + index * XLEN/8), and subsequently branch to the fetched address.  Two, cm.jt can branch to (jvt + index * XLEN/8) in a special execution mode where the next fetched XLEN bytes are considered to be an absolute jump instruction.  Strategy one represents fewer steps overall but requires indirect jump handling to be scheduled after data cache access, which will typically not be the case in single-issue in-order cores.  Strategy two minimizes pipeline impacts.

To make implementing strategy two in simple cores as straightforward as possible, cm.jt in the base ISA is defined to have the semantics of instruction fetches.  It requires X-permission, not R-permission, in the authorizing PTE or PMP, and is affected by incoherent instruction caches (requring fence.i after any software update to the jump table).

See riscv/riscv-code-size-reduction#195 for some relevant discussion.

sorear, 2024-01-27:
Advantages of using jvtc are that you can make a capability encompassing a jump table and all of the functions contained in the jump table, and then use it freely from multiple objects with fine-grained pcc bounds.  However, this has surprising security properties: since jvtc cannot be sealed, every object with access to the jump table has access to all globals that are accessible from functions in the jump table.  The jump table can still be used if none of its functions have access to sensitive data, which is likely to be the common case, but it makes it difficult for tooling to automatically construct a jump table without knowing which user functions have sensitive data.

Advantages of using pcc are that any subset of functions in a single object can be turned into a jump table for that object without changing security properties, which is a much closer match for how jump tables are actually used by the current tooling: the jump table is generated during the final link, used by the main object, and ignored by shared libraries.

I now favor pcc metadata, not jvtc metadata.

jrtc27, 2024-01-27:
A relevant observation to make here is that, conceptually, CM.JALT is designed as a way to further compress JAL / AUIPC+JALR, i.e. direct calls, not indirect ones, for which the CHERI equivalents still use PCC's bounds.

tariqkurd-repo, 2024-01-30:
Yes @jrtc27 is right - this is specifically for compressing JAL / AUIPC+JALR calls which don't update the PCC metadata, and so PCC metadata is the correct answer - and also the original intent which I will clarify in the doc.

Zcmt has a lot of future expansion space, so we can add other modes in the future if we can find interesting things to implement.

I'm glad we can agree on pcc bounds.

closed 2024-01-31 by andresag01