riscv/riscv-cheri #63: Save encoding space by using address LSB as mode bit

LawrenceEsswood, 2024-01-26:
We could save encoding space for whether a capability refers to code executing in cap mode or legacy mode by having this be indicated by the LSB of address. Note, this works well in conjunction with vectored exceptions whereas putting mode into the type field does not.

This would require having a new bit of state elsewhere. On trap, this bit could be recombined with PCC when storing to EPCC.

jrtc27, 2024-01-26:
Morello does this and it causes software pain. Putting it in the capability is a far more elegant solution rather than abusing the LSB for branch targets and storing the state separately in other situations.

sorear, 2024-01-27:
Care to elaborate?  I'm putting together my thoughts on Zcheri_legacy and "Zcheri_mode uses a reserved bit that might have better uses" currently figures highly on my list of reasons why one is better than the other.

arichardson, 2024-01-27:
Using the LSB of the address to indicate mode (like Morello/thumb) means software that looks at function addresses needs to be aware that the lsb has to be masked off before inspecting the target (e.g. unwinders or certain kernel functionality). It can also be annoying for the toolchain, especially the linker, but at least for llvm many issues have already been discovered and fixed. However when things go wrong it is really annoying to debug, I spent many hours fixing an incorrect mode on morello since the compiler decided to emit a relocation against section plus offset instead of the symbol and thus lost the mode information.

LawrenceEsswood, 2024-01-29:
The question is, is making life simpler (not making them have to mask) for unwinding logic really a good enough reason to consume an extra permission bit for every executable cap?

If we consume too many, we are locking ourselves into a corner. This is low hanging fruit that will be a lot harder to change later on.

tariqkurd-repo, 2024-04-04:
Hi @LawrenceEsswood - is this still relevant now that https://github.com/riscv/riscv-cheri/pull/155 has been merged?

arichardson, 2024-04-05:
I believe we can close this since we have a new encoding with mode.

closed 2024-04-05 by arichardson