riscv/riscv-cheri #64: Increase type space by splitting on X permission

LawrenceEsswood, 2024-01-26:
This idea lifted from CheriOT.

Types for capabilities tend to only make sense for either executable capabilities or non-executable capabilities. Having the X bit be an implicit low bit of a capabilities type (when sealed) doubles the type space.

This is very useful if we then want to put more information in type space rather than permission space.

sorear, 2024-01-27:
This can't be done as written since we don't have object types in this specification.

I filed #70 which should be equivalent to this once object types are added.

LawrenceEsswood, 2024-01-29:
Although there may be no way of constructing them now, this is still relevant to decide which type the single sealed (and executable) is for the purpose of cgettype.

If the future we want the low bit of a type to be inferred from x, the type should be chosen sensibly to fit this requirement.

tariqkurd-repo, 2024-04-04:
Hi @LawrenceEsswood - is this still relevant now that https://github.com/riscv/riscv-cheri/pull/155 has been merged?

arichardson, 2024-04-05:
This is still relevant but I think more as part of a "Zcheri_types" extension that adds more kinds of objects.

nwf, 2025-09-09:
> Can we close this one now that we have multiple roots in the specification?

This is more a question of _encoding_ than of _multiple roots_, I think.  Just to say, each of these are completely possible:
- a fully-orthogonal permissions encoding to nevertheless have multiple distinct roots (consider: W^X enforcement inductive from primordial conditions) and
- a non-orthogonal scheme with a single Infinite cap used as the one root everywhere.

Indeed, the present Appendix A RV32 AP field encodings -- with and without Zylevels1 -- are examples of the latter.

But, that said: those encodings use a 2-bit "quadrant" selector with one quadrant value biconditionally implying X permission, so this kind of a split has been in place for a while.

closed 2025-09-22 by arichardson