riscv/riscv-cheri #65: Save encoding space by putting ASR (and other mode switches) in type

LawrenceEsswood, 2024-01-26:
Also inspired by CheriOT as they did this for turning interrupts on/off.

Capabilities that should change some kind of mode (for example, whether CSRs can be accessed or not), can equally well be put into type rather than permissions.

When combined with splitting the type space this can save some bits.

More specifically, rather than just a SENTRY type, we can have SENTRY and SENTRY_ASR. Sealing with type SENTRY_ASR would require a capability with seal permission for that type.

sorear, 2024-01-27:
It is crucial in this case that the interrupt flag is part of architectural state and is accessible through a register after an exception (mstatus.MPIE).

pcc isn't a sealed capability and exceptions do not create a sealed capability, so the interrupt flag would be lost on exceptions otherwise.

If we adopted this, we'd need to add an PASR bit to track the effective ASR permission.  I am also unsure whether it is acceptable to require sentries for using ASR.

LawrenceEsswood, 2024-01-29:
I question whether EPCC should have always been a sentry on trap.  All the arguments about stopping accidental manipulation / use of CRA can be made about EPCC. If the kernel needs to skip an instruction, it can always unseal, which makes this manipulation of userspace control flow more explicit.

xret already unseals, I think?

But, yes, short of that we would need to track the bit on exception so that xret worked properly.

Also, it is not technically "required" to use sentries for ASR. A machine would start with ASR and never lose it unless a CHERI instruction were used.

tariqkurd-repo, 2024-04-04:
Hi @LawrenceEsswood - is this still relevant now that https://github.com/riscv/riscv-cheri/pull/155 has been merged?

arichardson, 2024-04-05:
Probably no longer needed with the current permissions format where ASR is only available for all-perms caps.

andresag01, 2024-10-01:
As mentioned above, this was already addressed because we have rules to save encoding space while keeping ASR as a permission. Also, the there has not been much activity in this ticket for months, so I will close it. Please create a new ticket if you'd like to continue the discussion.

closed 2024-10-01 by andresag01