riscv/riscv-cheri #67: Remove XLENMAX

sorear, 2024-01-26:
The ARC is [considering to make MXLEN constant](https://lists.riscv.org/g/tech-unprivileged/message/646).  If this happens, XLENMAX will become equal to MXLEN in all cases and we can simplify our spec by removing the Zcheri-specific term.

andresag01, 2024-01-29:
Agreed. It would be good to remove XLENMAX if the ARC accept that proposal.

arichardson, 2024-01-29:
[2.7.1 already states that MXLEN must be read-only ](https://riscv.github.io/riscv-cheri/#_machine_isa_register_misa)for Zcheri implementations, so maybe we can already drop XLENMAX?

andresag01, 2024-01-30:
> [2.7.1 already states that MXLEN must be read-only ](https://riscv.github.io/riscv-cheri/#_machine_isa_register_misa) for Zcheri implementations, so maybe we can already drop XLENMAX?

In the current spec, this is only the case for the Zcheri_purecap extension. Zcheri_legacy extension allows configuring MXLEN, SXLEN and UXLEN to any value as long as MXLEN >= SXLEN >= UXLEN in order to "maximize" compatibility with Standard RISC-V which supports that feature (see [here](https://riscv.github.io/riscv-cheri/#_machine_status_registers_mstatus_and_mstatush)), so XLENMAX != MXLEN in some cases.

tariqkurd-repo, 2024-01-31:
We've discussed this internally - the concern is that on ARC review that they will require writeable MXLEN. If they don't then we are very happy to remove it and have MXLEN==XLENMAX and remove all references to XLENMAX.

PR #123 (2024-04-05): Remove XLENMAX
As of riscv/riscv-isa-manual@86ff22d6 MXLEN is read-only and reflects the widest XLEN of the machine, so use that instead.

As of riscv/riscv-isa-manual@0d7621d9 MXLEN >= SXLEN >= UXLEN is required by the base privileged architecture, so it does not need to be restated here.

Resolves #67.
commit 4340db15 (2024-04-05): Remove XLENMAX (#123)

As of riscv/riscv-isa-manual@86ff22d6 MXLEN is read-only and reflects
the widest XLEN of the machine, so use that instead.

As of riscv/riscv-isa-manual@0d7621d9 MXLEN >= SXLEN >= UXLEN is
required by the base privileged architecture, so it does not need to be
restated here.

Also added a note on MXLEN requirements on privileged architecture version.

Resolves #67.
closed 2024-04-05 by andresag01