riscv/riscv-cheri #69: Define capability endianness

sorear, 2024-01-27:
An implementation of Zcheri where endianness is switchable at runtime must use a fixed endianness for capabilities, since RISC-V requires byte sequences* to remain invariant across endian switches, and if an invariant byte sequence were reinterpreted with the opposite endianness it would result in a catastrophic loss of provenance.

Such an implementation must present a capability endianness opposite its integer endianness in one mode, so a requirement that capability endianness equal integer endianness is unsupportable in the general RISC-V setting.  Allowing the capability endianness to be implementation defined separate from the integer endianness would contradict the requirement of an architectural capability representation that can be inspected in byte dumps.

As such: Require that capability representation be little-endian on all systems, or only define Zcheri for fixed little-endian systems.

* The precise wording of the requirement is that a byte written at some address will remain readable at this address.  Narrowly, this means that byte invariance is only required within a CLEN-bit region if the tag is zero, and an implementation could allow the capability endianness to appear to vary by reversing the bytes of CLEN-bit regions on loads and stores conditionally on a set tag.  This is unlikely to be microarchitecturally reasonable.

arichardson, 2024-01-29:
I agree that the representation needs to be fixed at runtime. Mandating little-endian seems like the most sensible thing to me. I'm not sure if Zcheri will ever exist on an bi-endian RISC-V implementation, but as a point of reference, Morello has the following constraints:
```
An implementation of the Morello architecture does not support the following:
• The AArch32 state.
• Mixed-endian at any Exception level.
• Fixed big-endian. The architecture only supports fixed little-endian
```

Also having worked with big-endian capabilities before (CHERI-MIPS), it is a lot nicer for software if the first quantity is the address as it allows code that simply fetches an address-size quantity (debuggers such as gdb/kernel backtracers/etc.) to work without additional changes.

jrtc27, 2024-01-29:
> Also having worked with big-endian capabilities before (CHERI-MIPS), it is a lot nicer for software if the first quantity is the address as it allows code that simply fetches an address-size quantity (debuggers such as gdb/kernel backtracers/etc.) to work without additional changes.

It’d have the right bytes for the address but they’ll be backwards. Only a true little-endian system lets you do that.

arichardson, 2024-01-29:
> It’d have the right bytes for the address but they’ll be backwards. Only a true little-endian system lets you do that.

Agreed, I should have been more clear on my comment that I meant little-endian caps on little-endian systems vs big-endian caps on big-endian systems.

tariqkurd-repo, 2024-01-30:
I'm happy with little-endian only like Morello

PR #119 (2024-02-19): Capabilities are little endian
Possible alternatives include requiring little endian for Zcheri_purecap, allowing bi-endian implementations to use an implementation defined endianness, and rules-lawyering the unprivileged ISA byte invariance requirement to only apply to CLEN-bit regions with the tag cleared.  This is the only option that is consistent with the architectural capability representation and how we handle variable XLEN.

Resolves #69.
closed 2024-02-19 by andresag01