riscv/riscv-cheri #70: Require X permission for CSEAL

sorear, 2024-01-27:
Sentry capabilities have, by design, no authority except control transfer, and a sentry capability with X=0 is useless except as a token.  If CSEAL is modified to unset the tag if the input lacks X permission, then the sentry case with X=0 becomes unconstructable and can be reused by future extensions.

jrtc27, 2024-01-27:
That requirement was deliberately dropped during the development of this spec (the CHERI spec enforces it) because people wanted tokens.

sorear, 2024-01-27:
I remain unconvinced that for the minimal first pass spec, applications can't simply use a RO capability to an otherwise unused subobject, or a 0-length capability to a _strictly_ interior address.

arichardson, 2024-02-09:
Since you can still unseal using CBUILDCAP, I believe having non-executable sentries is valuable. I guess you might be able to use a zero-perms/0-length capability to a compartment-local address and then rederive, but I find unsealing using BUILDCAP is a cleaner approach for tokens.

tariqkurd-repo, 2025-09-05:
https://github.com/riscv/riscv-cheri/pull/760 seems relevant

nwf, 2025-09-21:
#767 replaced #760; #795 is perhaps relevant, too (in that it allows capability encodings to use different otype values for X-granting and X-non-granting ambient seals).  I think this issue can be closed.