riscv/riscv-cheri #74: Should the main branch be deployed to GitHub pages on each push?

arichardson, 2024-01-29:
Right now https://riscv.github.io/riscv-cheri/ points to a manual workflow dispatch, but
deploying on each push would make it easier for people to read the latest specification without having to build it themselves. Maybe the manual workflow dispatch should just be needed to deploy a new PDF and create a release?

tariqkurd-repo, 2024-01-30:
my preference is to be a bit more controlled and choose when we make releases, e.g. after a few bug fixes have been rolled in.

arichardson, 2024-01-30:
I think this all depends on whether we view the HTML page as a release or just a latest snapshot. In my mind the PDFs were the release and HTML isn't, but I could definitely see both being treated equivalently. Happy to close this if you think we should treat them both as releases.

tariqkurd-repo, 2024-01-31:
I think that makes sense providing it's clear - HTML is the latest and the PDF is a release

closed 2024-02-06 by arichardson