riscv/riscv-cheri #80: Remove duplicate mnemonics

tariqkurd-repo, 2024-01-31:
CHERI is defined to have different mnemonics for capability mode and legacy mode. 
This has been controversial as it causes big changes in the assembler code and the software tools. It also has a big effect on the documentation, extending the document by many pages.

A change an assembly format is inevitable between the modes but it should be minimised.

This came up when discussing RISC-V integration with the RISC-V management - how much work is involved to port new extensions to CHERI? For Vector, for example, it is required to add _many_ new opcodes as there are so many loads and stores, and so this need to be simplified.

Therefore the proposal is as follows:

**All load/stores/atomics/CBOs/prefetches have the _same_ mnemonic in capability and legacy modes.**

currently we have e.g.

legacy mode: `ld rd, offset(rs1)`
capability mode: `cld rd, offset(cs1)`

the proposal is that the capability mode syntax will become:

capability mode: `ld rd, offset(cs1)`

so the operand type continues to be distringuished by `rs1/cs1` but the mnemonic remains the same.

I'm proposing to continue using `c` as the capability datatype so `lc/sc` etc.

`cincoffset[imm]` becomes `add[i]`

legacy mode: `add rd, rs1, rs2; add rd, rs1, imm`
capability mode: `add cd, cs1, rs2; add cd, cs1, imm`

`cmove` becomes `mv`

legacy mode: `mv rd, rs1`
capability mode: `mv cd, cs1`

`auipcc` becomes `auipc`

legacy mode: `auipc rd,  imm`
capability mode: `auipc cd, imm`

`cjalr` and becomes `jalr`

legacy mode: `jalr rd, rs1, offset`
capability mode: `jalr cd, cs1, offset`

and we no longer need `jalr.cap/jalr.pcc` in the assembler, they can be inferred by

legacy mode: `jalr cd, cs1, 0`
capability mode: `jalr rd, rs1, 0`

i.e. zero offset and referring to the data-type from the _other_ mode

`Store conditional`

`sc` -> store cap
`sc.<type>` -> store conditional of <type> which can be `c`, so `sc.c` is valid for store conditional of a capability

sorear, 2024-01-31:
Strong agree for memory access instructions.  Less sure about overloading add/addi/mv.

The move from RV32 to RV64 sets the precedent of reusing mnemonics for the "same" operation on a different size of data, but where there are two different operations, like add or addw, they are distinguished by mnemonic.

Is it OK for "jal symbol" and similar pseudoinstructions to access pcc and cra without naming either capability?

andresag01, 2024-01-31:
I agree that unifying most of those mnemonics is a good idea.

However, I am concerned about cincoffset vs add. cincoffset does not have the same encoding as add, so these are in fact different instructions and their mnemonics must be different.

tariqkurd-repo, 2024-01-31:
`cincoffset` vs `add` is the same argument as `add` vs `addi` where they are distinguished by operand type

`add rd, rs1, rs2`
`add rd, rs1, imm` <-- different opcode

the big motivation behind renaming `cincoffset[imm]` to `add` is to simply show the path from (e.g.) old function prologues and epilogues. it is, after all, an `add`.

for RV32:

```
add sp,sp,-96
sw ra,92(sp)
sw s0,88(sp)
sw s1,84(sp)
```
becomes

```
add csp, csp, -192
sc ra,184(csp)
sc s0,176(csp)
sc s1,168(csp)
```
so the code looks very familar - that's the intention.

we can of course choose another mnemonic for `cincoffset[imm]` but it _must not_ look alien like the current one.

>Is it OK for "jal symbol" and similar pseudoinstructions to access pcc and cra without naming either capability?

in capability mode than `jal symbol` will indeed link to `cra` if it's not named and assumed.

jonwoodruff, 2024-01-31:
Not really adding much, but throwing in my opinion/observation:

If we follow the rv32->rv64 change in base ISA:
`l(bhwd) rd, rs(##)`
Is brought right over unchanged; the implied type of rs grows from 32 bits to 64 bits as that is the only address size supported in that mode.  If we strictly followed that lead, we wouldn't even change the name of the register operand.

`add rd, rs1, rs2`
Changed the inferred type to the new xlen, and we grew a `addw` for the legacy data type.  If we're explicitly keeping xlen=64, then maybe the most sensible thing is to grow an `addc` for clen?

`jalr rd, rs`
Changed the implied PC width without changing the mnemonic, as xlen addresses (including PC addresses) are all that exist in the new base ISA.  Following that pattern, we would just reuse `jalr rd, rs` directly.

I don't actually know what the advantage of using a different register alias (`c# ` rather than `r#`) is for purecap.  I guess it's a bit alien to risc-v?  Though obviously not strange from an x86 perspective.  I like the explicitness of it, but we're used to different instructions operating on different widths of the same register name (`addw` vs `add`, or `sb` vs `sd`).

tariqkurd-repo, 2024-01-31:
using `c` for registers is alien, thats's true, but I think it makes sense

`ld rd, offset(rs1)`
`lc cd, offset(rs1)`
`lc cd, offset(cs1)`

all do different things, and I think they're all quite clearly specified by the syntax.

and I think `addc` is ok as a mnemonic.

do you think this makes sense @jonwoodruff 

legacy mode: `jalr cd, cs1, 0`
capability mode: `jalr rd, rs1, 0`

so we can be explicit about which type we are jumping to and linking to?

arichardson, 2024-01-31:
Thanks for writing this up @tariqkurd-repo! This has come up in many discussions over the past years and I much prefer the Morello approach of the instruction being determined by the register argument over a prefixed instruction. Since the type of operation is still indicated by the register we also don't lose intentionality and risk assembly code that is not ready for CHERI being accepted by the compiler without the required changes.

Since the load/store instructions are never both available to the assembler it is also unambiguous even if we were to reuse rs1 vs cs1, but I think being explicit that we are using the C registers is much nicer.

Regarding `addc`, that mnemonic is sometimes used for operations with using the architectural carry bit (e.g. [Arm ADDC](https://developer.arm.com/documentation/101655/0961/8051-Instruction-Set-Manual/Instructions/ADDC)), so it could be confusing to developers coming from the Arm ecosystem but it's probably fine. The C suffix is also less confusing since it does not look similar to the compressed `c.add`.

I also think having the `jal`and `ret` pseudos being mode dependent aliases would also be nice since it reduces developer friction.

jrtc27, 2024-01-31:
One thing to bear in mind here is that LLVM's assembly parser assumes that the instruction can be uniquely determined from the mnemonic and first operand, which wouldn't be true in this case, and so needs hacks to make it work. It can be done though, as evidenced by Morello, and LLVM's internals today shouldn't overly influence ISA design, they're mostly a reflection of what has historically been required.

jrtc27, 2024-01-31:
I personally wouldn't add the C suffix for ADDC though, that really does risk confusion (though it's ADC in x86 and our CHERI-x86 sketch has ADDC be capability add with an explicit operand size...). The reason you need ADDW is that you can't name the 32-bit sub-register, but we can disambiguate between X and C, so don't need such a suffix.

sorear, 2024-01-31:
> cincoffset vs add is the same argument as add vs addi where they are distinguished by operand type
> 
> add rd, rs1, rs2
> add rd, rs1, imm <-- different opcode

I and at least several others regard the second form as a GNU binutils bug which is preserved for compatibility.  You will not find add with immediate in any RISC-V specification, both the ISA manual and the assembly programmer's manual use

> addi rd, rs1, imm

exclusively.

That said, the topic of this bug is "Remove duplicate mnemonics".  I don't see anyone here objecting to removing the duplicate mnemonics for instructions that keep the same encoding and behavior, so why not focus on the part with consensus here and have a separate issue for assembly-level register intentionality?

jrtc27, 2024-01-31:
> I and at least several others regard the second form as a GNU binutils bug which is preserved for compatibility. You will not find add with immediate in any RISC-V specification, both the ISA manual [...]

Because none of them are duplicate mnemonics, they are different instructions that share the same encodings and are very closely related. It's all just shades of the same question.

tariqkurd-repo, 2024-02-01:
OK - to summarise -

1. we are proposing dropping the `c` prefix from load/stores/atomics/CBOs, so e.g.

legacy mode: `lw rd, offset(rs1)`
cap mode: `lw rd, offset(cs1)`

2. we are proposing keeping `c` as the data type for load/stores/atomics/CBOs, e.g.

legacy mode: `lc cd, offset(rs1)`
cap mode: `lc cd, offset(cs1)`

3. we are proposing distinguishing between  `jalr` types by operand

legacy mode: `jalr cd, cs1, 0; #currently jalr.cap`
legacy mode: `jalr rd, rs1, offset; #currently jalr`

capability mode: `jalr rd, rs1, 0; #currently jalr.pcc`
capability mode: `jalr cd, cs1, 0; #currently cjalr`

`jalr.pcc/jalr.cap` does need a name for the mnemonic, even if it's not required by the assembly syntax. Any suggestions? We could keep `jalr.pcc/jalr.cap` for the mnemonics and just update the assembly syntax

4. `cincoffset[imm]` need to map to something including the word `add`

`add cd, cs1, rs2`
`addi cd, cs1, imm`

seem acceptable? the mnemonic still needs a name though, maybe `addcap/addcapi`?

arichardson, 2024-02-09:
Once this renaming has been completed we can also delete most of "Table 43. Mnemonics with the same encoding but mapped to different instructions in Legacy and Capability Mode"

tariqkurd-repo, 2024-02-12:
>Once this renaming has been completed we can also delete most of "Table 43. Mnemonics with the same encoding but mapped to different instructions in Legacy and Capability Mode"

that table is auto-generated in python so the contents mostly will just disappear as I progress with the renaming.

tariqkurd-repo, 2024-02-14:
https://github.com/riscv/riscv-cheri/pull/109
merged

closed 2024-02-14 by tariqkurd-repo