riscv/riscv-cheri #83: CSeal does not clear tag if the capability is already sealed

francislaus, 2024-02-01:
`CSeal` differs from `CSealEntry` in its behaviour. `CSeal` allows "endless sealing" if the capability is already sealed. In contrast, `CSealEntry` will clear the tag if the capability is already sealed.

tariqkurd-repo, 2024-02-01:
Presumbly this prevents them being equivalent with the 1-bit otype, and so we need to choose the correc tone so I'll mark this as a `bug`.

arichardson, 2024-02-02:
Clearing the tag for already sealed capabilities in csealentry was only needed since the object type could have been different in which case sealing as a sentry would actually be non-monotonic. If I recall correctly the reason I didn't special case sentries as a valid no-op was just to catch programming bugs (the same reason as requiring execute permissions). I don't really mind whether we allow or prohibit the no-op seal.

francislaus, 2024-02-02:
@arichardson As you described, there is no problem with monotonicity. However, I wonder what would be the most approach for software in the long term. If otypes (even in a very limited form) are added to the CHERI RISC-V spec, we would need to change the behaviour of this instruction. Is this a valid route in the ratification process of an extension to CHERI RISC-V?

andresag01, 2024-02-02:
I agree with @francislaus : It would be good to think what would be best in the long term if, for example, something like otypes were to be brought in. If there are more times, is this instruction still CSeal or CSealEntry? The conclusion from sometime ago was that we should either rename this instruction to CSealEntry, in which case it would be good to clear tag if sealed, or keep it as CSeal but potentially allow another operand to indicate the type.

I would be in favour of just making the current encoding CSealEntry because the future requirements are still unclear.

francislaus, 2024-02-06:
Changing `CSeal` to be `CSealEntry` seems to be the most sensible route forward to me. We are not yet sure in which shape - or if at all - otypes will appear in the spec.

tariqkurd-repo, 2024-02-07:
ok - so let's replace `CSeal` with `CSealEntry` and keep the existing semantics of `CSealEntry`. PR required!

andresag01, 2024-02-09:
https://github.com/riscv/riscv-cheri/pull/105 fixes this.

closed 2024-02-09 by andresag01