riscv/riscv-cheri #84: Tag memory PMA would be useful

tariqkurd-repo, 2024-02-01:
In a real CHERI system it is useful to distinguish memory which is tag aware and memory which is not - i.e. doesn't support tags.

This will allow a CHERI core to mark transactions as `tag aware` or `non-tag aware`.

It seems reasonable that all cached memory is `tag aware`, but this may not be a firm requirement.

Therefore I propose adding a new PMA attribute for this, with a recommendation that cached memory is tag aware.

tariqkurd-repo, 2024-02-02:
@andresag01 has pointed out that this is included already.
with options that LC/SC with the tag set can either cause an access fault exception, or store the tag as zero.

I think clearing the tag is useful in both directions (load and store) so that LC/SC can be used for performance. Therefore they become double width load/store which don't affect the tag state.

So I think that the spec should be updated to include loading tags as zero as an option.

andresag01, 2024-02-02:
Yes, there is currently a short section in Chapter 3. This is what it says:

> Physical Memory Attributes (PMA)
Typically, the entire memory space need not support tagged data. Therefore, it is desirable that harts
supporting Zcheri_purecap extend PMAs with a taggable attribute indicating whether a memory
region allows storing tagged data.
When the hart attempts to store or load data with the tag set to memory regions that are not taggable,
the implementation may:
• Cause an access fault exception
• Implicitly set the stored tag to 0

Yes, sorry, I missed the load-side of this.

andresag01, 2024-02-02:
@tariqkurd-repo: Please note that the load part is being fixed here: https://github.com/riscv/riscv-cheri/pull/54/files#diff-a99e635c97472fffc8cbe6ae563b95dceb0eb931705c6fa6d62df7ce586c981aR1096

tariqkurd-repo, 2024-02-02:
yes - this was fixed by [https://github.com/riscv/riscv-cheri/pull/54](https://github.com/riscv/riscv-cheri/pull/54/files#diff-a99e635c97472fffc8cbe6ae563b95dceb0eb931705c6fa6d62df7ce586c981aR1096)

closed 2024-02-02 by tariqkurd-repo