riscv/riscv-cheri #85: Checking whether PCC is sealed

francislaus, 2024-02-02:
The spec says that for each instruction PCC must be checked for a set of conditions. One is whether PCC is not sealed. I believe that none of the instructions in the current form of the spec can bring a sealed capability into PCC. If we can convince ourselves of this invariant being true at all times, we could drop this architecturally enforced checked on each instruction. This might be a property that can be confirmed by Sail proofs.

andresag01, 2024-02-02:
There is a contrived case, which might actually happen in the current spec, when the hart goes to debug mode and the debugger writes a sealed capability to `dpcc` then executes `dret`. In this case, the spec does not require that `dpcc` is unsealed before writing to `pcc`.

But I think you are right: it should not be possible to write a sealed capability to `pcc`. I think we should:
* Tidy up the debug part of this
* Remove that check from `pcc` and add a note explaining the rationale for not including that check

Hopefully there is a way to prove that `pcc` will never be "sealed", if not, then it would be prudent to keep the check in.

tariqkurd-repo, 2024-02-13:
`dpcc`,  `mepcc` and `sepcc` all unseal on execution
so i think Franz is right - pcc can never be sealed.