riscv/riscv-cheri #86: Alignment requirements of CLC and CSC instructions

francislaus, 2024-02-02:
The spec says that CLC and CSC need to be aligned to `CLEN/8`. Looking at the unprivileged base ISA spec, it seems that load and store instructions have no misaligned address exceptions. This is handled via the EEI. We had an internal discussion at Cambridge and we do not think that we should enforce alignment in the architecture. However, a misaligned CLC and CSC should be tag-stripping.

https://github.com/riscv/riscv-cheri/blob/90683c3b5448d90ece38febf7326dac8fb6aca64/src/insns/load_exceptions.adoc?plain=1#L6-L8

andresag01, 2024-02-08:
As discussed today, the existing RISC-V load/store instructions support misaligned accesses in hardware if the `Zicclsm` extension is implemented, so ideally load-cap/store-cap instructions will follow the same pattern. We should change the spec accordingly.

andresag01, 2024-02-08:
@arichardson : Any thoughts on making load-cap/store-cap not trap on misaligned accesses when the implementation supports `Zicclsm`?

jrtc27, 2024-02-08:
What’s the justification?

tariqkurd-repo, 2024-02-09:
The idea is to allow LC/SC to be used more flexibly to make use of the double width load/store bandwidth for performance - so that if they are misaligned and/or C permission isn't granted than all tags are stripped/cleared. If they are aligned and with C permission then the access the memory tags as expected.

jrtc27, 2024-02-09:
The only time that’s useful is if you know your region *doesn’t* contain capabilities. Otherwise you have to take care to use aligned accesses to preserve tags. It’s very hard to actually use that feature, and I am very concerned it will lead to broken code being written. I just do not think it’s useful in practice. Capability alignment isn’t required by the instructions for hardware limitation reasons, as is the reason for integer alignment requirements, it’s required for sensible code reasons, so it’s a very different situation to Zicclsm.

I would want to see a concrete legitimate use for this before being happy to allow this.

arichardson, 2024-02-09:
I am concerned about relaxing that requirement.
If we compare this to atomic, they require alignment since silently doing a non-atomic operation would be dangerous. While tag stripping is not necessarily dangerous, it is almost certainly a programmer error, so should result in a trap.

Maybe a load without PERMIT_LOAD_CAP is legal, but I'm not sure how useful that kind of optimization will be since you'd need to strip the permission before the load/store at which point you might as well do two XLEN loads/stores. If someone has a real example of this improving codegen I could be convinced it's something we should support.

PeterRugg, 2024-02-13:
> it is almost certainly a programmer error, so should result in a trap

That sounds like exactly the reasoning we used to have for all of the other instructions throwing traps, e.g. `CSetBounds` when widening bounds, etc. We moved away from that for various reasons. However, since loads and stores are so obviously side-effecting and have to throw exceptions anyway, I suppose most of those reasons (e.g. allowing code-hoisting) don't apply. It also seems like tags disappearing from a faulty `memcpy` are likely to go unnoticed for much longer than in capability arithmetic cases.

On another note, the RISC-V spec says:
> An EEI may guarantee that misaligned loads and stores are fully supported, and so the software running inside the execution environment will never experience a contained or fatal address-misaligned trap.

With the current proposed behaviour, this is no longer true. I believe that the intent is for address misaligned traps to be handled by firmware to support this: might we want to allocate a different (CHERI) exception code for this. As @jrtc27 points out we're trapping for a different reason than the existing misaligned trap, and it might help firmware to differentiate the two cases more easily.

jrtc27, 2024-02-13:
It’s not very useful to give it a different code. Trap handlers already need to decode the instruction if they want to emulate it, so all you do is redirect the trap (code that already exists for generic traps) if it’s an instruction that can’t be emulated, like this. It’s appropriate to use the misaligned load/store exceptions for a misaligned load/store of a capability, and doing something else would be confusing and introduce an unnecessary new exception.

andresag01, 2024-02-16:
As discussed, capabilities are always aligned to CLEN/8 bytes, so attempting to use load/store capability with a misaligned address is almost certainly a software error. Raising a misaligned access exception in those cases helps debugging. Therefore, we will keep the current behaviour for these instructions which is to except on misaligned access.

There is already very strong wording for this in the specification (see [here](https://github.com/riscv/riscv-cheri/blob/main/src/riscv-integration.adoc?plain=1#L169)), so I think we can close this ticket. What do you think @tariqkurd-repo ?

tariqkurd-repo, 2024-02-16:
I think a note in the spec would be useful, to prevent this coming back again (or if it does to give a reason why it was rejected)

closed 2024-02-23 by tariqkurd-repo