riscv/riscv-cheri #90: prefer AUIPC over CSRR for debug pcc access

sorear, 2024-02-06:
First, I hope we can agree that `csrr rd, pcc` and `auipc rd, 0` have exactly the same architectural effect in all execution modes in which they are supported, so any core which implements one can implement the other with, at worst, decoder changes.  The only reasons to prefer one over the other are opcode space allocation and naturalness for various implementation strategies.

Opcode space favors `auipc` - pcc as a CSR represents 6*1024 opcodes, of which 128 are redundant with cases of `auipc` and the rest are always illegal.

Rocket and NOEL-V provide the pc to the CSR write procedure solely for the purposes of validating `misa.C` writes.  BOOM does not even have the pc as part of the micro-op; CSR writes only have access to the pc as a side effect of the fact that BOOM currently implements all CSR instructions as machine-serializing, so the pc of the CSR instruction is the same as the ROB commit pc.

I can't think of any implementation strategy which would preclude providing an arbitrary value of pcc with known metadata to auipic instructions during debug halt, but which would allow CSR reads of pcc under the same conditions.  Please advise.

Another option would be for x0 to read as an Infinity capability in debug mode; this would prevent `hartinfo.dataaccess=1` (data registers shadowed within 2 KiB of x0) from being useless in a CHERI environment.

PeterRugg, 2024-02-06:
> First, I hope we can agree that `csrr rd, pcc` and `auipc rd, 0` have exactly the same architectural effect in all execution modes in which they are supported

I'm not sure I agree? This prompted me to consider what `CSRR rd, pcc` means and raise https://github.com/riscv/riscv-cheri/issues/92. Regardless, when `CSRR cd, pcc` does exist in a way that reads the full capability, it should do so in legacy mode, whereas `AUIPC rd, 0` should only get the metadata in capability mode (when it is in fact `auipcc cd, 0`)?

sorear, 2024-02-06:
Ignore Legacy mode for now - I'm trying to separate the Legacy mode and debug mode uses of the `pcc` CSR name.

andresag01, 2024-02-14:
Stepping back a little, the purpose of PCC in a CSR is for debuggers to get a hold of its value.

Recall that entering debug mode causes PCC to be set to Infinite capability. And that debuggers will need a source of authority, for example, to perform memory access particularly when only implementing Zcheri_purecap and there is no ddc. In this case, you need to get your source of authority from somewhere and a way is to provide access to PCC which is set to Infinite when entering Debug mode. However, the problem is that the debug spec allows implementations to not support executing PC relative instructions from the program buffer (like AUIPCC) and this is a very common implementation choice. So the solution currently in the spec is to expose PCC as a CSR so the debugger can simply do `csrr cd, pcc` to get that source of authority.

There are a couple of alternative ways (inspired from @arichardson comment [here](https://github.com/riscv/riscv-cheri/issues/92#issuecomment-1936556558)) to provide that source of authority:

1. Add a debug-only instruction, like `dret`, that outputs Infinite cap when executed (e.g. DGETINFINITECAP cd)
2. Add a new read-only, debug-only CSR number that always return infinite cap when (read instead of PCC!)

My preference is (2) because its generally easier to add CSRs than instructions. What do you think @sorear @arichardson @PeterRugg ?

tariqkurd-repo, 2024-02-14:
See https://github.com/riscv/riscv-cheri/pull/113

PeterRugg, 2024-02-14:
I agree, (2) is fine: it seems neater to have an always-almighty as opposed to shoehorning it into PCC. Note that the original reason that PCC existed in ISAv9 was to facilitate mode switching from legacy mode, but that's no longer an issue, as discussed elsewhere.

sorear, 2024-02-16:
(1) and (2) are both fine for me.  (I have an infinitesimal preference for (1) due to less opcode space usage.)

sorear, 2024-02-16:
Another idea would be to make `c0` read as Infinity in debug mode; rocket's implementation uses `dataaccess=1` and depends internally on being able to access memory relative to `x0` as part of the Debug ROM.

closed 2024-02-20 by tariqkurd-repo