riscv/riscv-cheri #92: No way to achieve `CSRR` on a capability

PeterRugg, 2024-02-06:
The RISC-V spec says that `CSRR rd, csr` is encoded as `CSRRS rd, csr, x0`. Under the current CHERI-RISC-V spec, `CSRRS` can only read the address, not the metadata. `CSRRW` has a special case for `rd=x0` but none for `rs1=x0`, so unless I'm missing something, it's not possible to read the full value of a capability CSR without overwriting its contents?

andresag01, 2024-02-06:
Yes, you are correct, the RISC-V spec states this:

> For both CSRRS and CSRRC, if rs1=x0, then the instruction will not write to the CSR at all ...
Both CSRRS and CSRRC always read the addressed CSR and cause any read side effects
regardless of rs1 and rd fields. Note that if rs1 specifies a register holding a zero value other
than x0, the instruction will still attempt to write the unmodified value back to the CSR and
will cause any attendant side effects. A CSRRW with rs1=x0 will attempt to write zero to the
destination CSR.

This needs to be fixed, otherwise there is no way to read a whole cap from CSRs without writing to it.

arichardson, 2024-02-06:
Reading the priv spec, it sounds like we have to change CSRS (and most likely also CSRC) to return the full capability register in capmode. This then allows `csrr cd, <reg>` to be encoded as `csrs cd, <reg>`. I don't think this should add much implementation complexity?

jrtc27, 2024-02-06:
The main issue here would be needing to special-case things so that if you're trying to read a sealed capability it doesn't count as a set address

arichardson, 2024-02-06:
> The main issue here would be needing to special-case things so that if you're trying to read a sealed capability it doesn't count as a set address

The privileged spec already has the following wording:

> For both CSRRS and CSRRC, if rs1=x0, then the instruction will not write to the CSR at all, and so shall not cause any of the side effects that might otherwise occur on a CSR write, nor raise illegalinstruction exceptions on accesses to read-only CSRs. Both CSRRS and CSRRC always read the addressed CSR and cause any read side effects regardless of rs1 and rd fields. Note that if rs1 specifies a register holding a zero value other than x0, the instruction will still attempt to write the unmodified value back to the CSR and will cause any attendant side effects. A CSRRW with rs1=x0 will attempt to write zero to the destination CSR.

This implies that using x0 is just a read?

PeterRugg, 2024-02-08:
Hmm, thinking on this a little more: isn't the main purpose of having `pcc` in a CSR to allow legacy mode (I guess without fast mode switch instructions) to get hold of a PCC to jump to to get itself into cap-mode in the first place? Wouldn't making CSRRS only return the full capability in cap mode undermine that goal?

arichardson, 2024-02-09:
> Hmm, thinking on this a little more: isn't the main purpose of having `pcc` in a CSR to allow legacy mode (I guess without fast mode switch instructions) to get hold of a `pcc` to jump to to get [...]

I agree that without the mode switch you would need an explicit instruction to get PCC, so allocating csr space for it doesn't make much sense IMO. But isn't that an orthogonal issue? This issue is about being able to read a CLEN-size csr without writing to it and making csrrs mode-aware seems like the approach that makes the most sense.

jrtc27, 2024-02-09:
Is there actually a sensible use case for having legacy support without a mode switching instruction?

arichardson, 2024-02-09:
> Is there actually a sensible use case for having legacy support without a mode switching instruction?

I doubt it and requiring the switch would also avoid the need for this kind of getpcc workaround.

jrtc27, 2024-02-09:
> > Is there actually a sensible use case for having legacy support without a mode switching instruction?
> 
> I doubt it and requiring the switch would also avoid the need for this kind of getpcc workaround.

Unless there's a compelling reason to support such a configuration it seems like optionality for the sake of it rather than because it serves a purpose.

PeterRugg, 2024-02-09:
To be clear then, we're proposing that Zcheri_mode (or whatever it ends up getting called) is mandatory if you support both the new base ISA (Zcheri_purecap) and Zcheri_legacy? I suppose it must remain a separate extension then to allow Zcheri_purecap only or Zcheri_legacy only to exist. Regardless, sounds like mode issues are being discussed elsewhere, so I'll submit a PR to fix this that just gets the capability metadata in capmode.

tariqkurd-repo, 2024-02-09:
This removes the point of Zcheri_mode entirely - and the simple objective here was to allow legacy mode without allocating the mode bit in the PCC - so that it can be used for another purpose.

We could argue that this isn't useful - merge Zcheri_legacy and Zcheri_mode - as they previously were before I separated out Zcheri_mode - and say that if that `pcc.M` is reserved it must be a purecap only core.

This is certainly possible as a direction, and a specification simplification.

andresag01, 2024-02-09:
> Unless there's a compelling reason to support such a configuration it seems like optionality for the sake of it rather than because it serves a purpose.

I answered a similar question in the CHERI SIG's mailing list, so I will paste some of that answer here:

Compatibility with standard RISC-V is a key goal of the currently proposed CHERI draft specification with the expectation that this will facilitate ratification. Another equally important goal was that the specification allows designing purecap-only implementations. Therefore, the specification was split into three main extensions.

- **Zcheri_purecap:** The core of CHERI which is sufficient for purecap-only implementations

Then there are two extensions providing different levels of support for compatibility with standard RISC-V:

- **Zcheri_legacy:** Introduces Legacy mode. The machine's current CHERI execution mode is given by a bit in xenvcfg. The idea is that supporting Zcheri_purecap and Zcheri_legacy is sufficient for implementations that are expecting to execute programs that are either only purecap, or only capability unaware standard RISC-V. Zcheri_legacy by itself is not meant for hybrid software that expects to frequently change CHERI execution mode, for example, a program that is mostly purecap but relies on a library that is capability unaware standard RISC-V. Therefore, Zcheri_legacy does not allocate a mode bit in the capability metadata, so there are more reserved bits in the capability format for future extensions -- this is particularly important for RV32 where space is very limited.
    - Note: M-mode must still be capability-aware because there is currently no way of forcing it to Legacy with Zcheri_legacy only in the current spec.

- **Zcheri_mode:** Built on top of both Zcheri_purecap and Zcheri_legacy. The extension introduces the mode (M) bit in the capability format and instructions to switch CHERI execution mode. Implementations supporting Zcheri_mode are suitable for executing hybrid programs efficiently. Implementations supporting Zcheri_mode are of course able to execute purecap and capability unaware standard RISC-V programs as well.

jrtc27, 2024-02-09:
Ok, so the use case is you want to have a CHERI system and run some old binaries without needing to allocate something in PCC’s metadata to support it. So in that world there’s no need to read PCC in legacy code and this the concern doesn’t exist?

andresag01, 2024-02-09:
> I agree that without the mode switch you would need an explicit instruction to get PCC, so allocating csr space for it doesn't make much sense IMO. But isn't that an orthogonal issue? This issue is [...]

The purpose of PCC in a CSR is not for programs in Legacy mode. It is for debuggers to get a hold of PCC.

Recall that entering debug mode causes PCC to be set to Infinite capability. And that debuggers will need a source of authority to, for example, perform memory access particularly when only implementing Zcheri_purecap and there is no `ddc`. In this case, you need to get your source of authority from _somewhere_ and the easiest way is to provide access to PCC which is set to Infinite. However, the problem is that the debug spec allows implementations to not support executing PC relative instructions from the program buffer (like AUIPCC) and this is a very common implementation choice. So the solution currently in the spec is to expose PCC as a CSR so the debugger can simply do `csrr cd, pcc` to get that source of authority.

andresag01, 2024-02-09:
> Ok, so the use case is you want to have a CHERI system and run some old binaries without needing to allocate something in PCC’s metadata to support it. So in that world there’s no need to read PCC in legacy code and this the concern doesn’t exist?

@jrtc27 : Yes. And the purpose of PCC access via a CSR is as I described above

andresag01, 2024-02-09:
Could we change the spec as follows to resolve this issue?
* CSRRS/CSRRC, if rs1=x0, to only _read_ a full capability when the CSR number accessed is a CLEN-wide CSR
* CSRRS/CSRRC, if rs1!=x0, _write_ the address part of the CSR in the way the instruction normally would with the CHERI constrains that are well understood (rep, sealing, etc check) and:
    * _read_ XLEN (i.e. address only) if the CSR number accessed is a XLEN-wide CSR
    * _read_ CLEN (i.e. whole cap) if the CSR number accessed is a XLEN-wide CSR.

arichardson, 2024-02-09:
> Could we change the spec as follows to resolve this issue?
>
> * CSRRS/CSRRC, if rs1=x0, to only _read_ a full capability when the CSR number accessed is a CLEN-wide CSR
> * CSRRS/CSRRC, if rs1!=x0, [...]

I agree with this, but for the rs1=0 case it should still read XLEN for an XLEN-wide CSR?

For clarification, are you proposing it reads CLEN regardless of mode or only when in capability mode?

arichardson, 2024-02-09:
> The purpose of PCC in a CSR is not for programs in Legacy mode. It is for debuggers to get a hold of PCC.
>
> Recall that entering debug mode causes PCC to be set to Infinite capability. And that [...]

If it's just for the debugger, wouldn't it also be simpler to allocate a `dinfinite` CLEN CSR (or some more sensible name) to get access to inifinitey instead of writing PCC as infinite and then reading that?

andresag01, 2024-02-12:
> I agree with this, but for the rs1=0 case it should still read XLEN for an XLEN-wide CSR?
>
> For clarification, are you proposing it reads CLEN regardless of mode or only when in capability mode?

I am proposing that these instructions do not depend on the mode. The amount of data you read/write depends on the CSR number that you access just like in the case for CSRRW:

* If you access any of the standard RISC-V CSR numbers, you will always get an XLEN-wide read
* If you access any of the new alias CSR numbers from the CHERI extension, you will always get a CLEN-wide read

> If it's just for the debugger, wouldn't it also be simpler to allocate a dinfinite CLEN CSR (or some more sensible name) to get access to inifinitey instead of writing PCC as infinite and then reading that?

That's another way of doing it and that register would need to only be accessible in Debug mode.

PR #108 (2024-02-14): Allow CSRRS/C to read CLEN-wide values
Fixes #92
closed 2024-02-14 by arichardson