riscv/riscv-cheri #93: Accessing CSRs without ASR permission

francislaus, 2024-02-06:
The current form of the spec says that "All non-user mode accessible CSRs require ASR-permission, including existing RISC-V CSRs." We were wondering whether that should hold for all non-user mode accessible CSRs. For example, we think that the `cycle` CSR should be readable on all privilege levels regardless of the ASR-permission.

francislaus, 2024-02-06:
This is the list of CSRs our cores allow to read:

`FFLAGS`
`FRM`
`FCSR`
`CYCLE`
`TIME`
`INSTRET`

arichardson, 2024-02-06:
Should this be extended to all performance counter CSRs? S-mode can still disable access for U-mode using `scounteren`? Requiring ASR here seems a bit restrictive. Alternatively, if we are really worried about using them for timing side-channels in compartmentalized programs, there could be a CSR bit that defines whether HPM access needs ASR.

andresag01, 2024-02-07:
@francislaus : Thanks for pointing this out. The intention is that all registers that are unprivileged according to the RISC-V privileged ISA remain accessible regardless of the ASR permission. This is the list I am talking about from the RISC-V spec (see attachement).
[image]

Ideally the CHERI spec uses the same language that the RISC-V privileged spec does, so perhaps it is best to modify this sentence to say _privileged CSRs_ instead of _non-user mode accessible CSRs_.

Does that make sense? Would you like to create a PR to address this wording issue?

arichardson, 2024-02-09:
I think it would be good to align the ASR behaviour with Table 2.2 in the RISC-V spec. The current sail-cheri-riscv model does not confirm to that behaviour, hopefully the new version for this spec will have it?

francislaus, 2024-02-14:
I think this raises the question on whether all CSRs are of the same privilege level. For example, reading a counter register does not provide you with the same privilege as being able to write to `mepc`. Code running in M mode that only wants to have access to counters would need the ASR bit set and thus would have more privileges than it should have.

tariqkurd-repo, 2024-04-04:
@francislaus are you suggesting that ASR should become more than one permission? M-ASR for M-mode state e.g. `mepc`, and S-ASR for S-mode state?

jrtc27, 2024-04-04:
I'd say no, with the usual "that breaks virtualisation" argument.

PR #192 (2024-04-09): Clarify non-user mode CSRs in CSR insns listing
Fixes #93
commit 54ea1ec1 (2024-04-09): Clarify non-user mode CSRs in CSR insns listing (#192)

Fixes #93
closed 2024-04-09 by andresag01