riscv/riscv-cheri #106: Renaming ASR permission

arichardson, 2024-02-09:
As has been noted before, ASR grants additional privilege beyond CSR access so the historical "Access System Registers" name may not be the best. Morello uses System which is IMO better, but the abbreviation would conflict with `S` for Store, maybe `Privileged Permission (P)`/`Elevated Permission (E)`/`System Management Permission (SM)`?

tariqkurd-repo, 2024-02-13:
What should the spec of ASR be? What additional state does it allow access to over CSRs?

arichardson, 2024-02-14:
I guess you could argue that instructions like `xret` actually read the CSR so ASR isn't needed for the instruction itself but rather the side-effects of that instruction. We can probably ignore this until there is another non-CSR case that needs elevated privilege.

I do wonder if we should require it for ECALL though...

sorear, 2024-03-11:
#140 and https://lists.riscv.org/g/sig-cheri/message/68 have a fair amount about other uses for ASR and its fundamental nature as the catch-all for resources, whether architectural or software-defined, accessible in a given privilege mode for which a more specific permission is either not needed or intended to be defined at a higher level using sentries.

Personally of the options given I prefer "Elevated (E)" - Privileged invites confusion with privilege levels, and System Management has implications about resource scope which may not be accurate.

tariqkurd-repo, 2024-04-04:
@arichardson do you want to do something about this one? Or shelve it?

andresag01, 2024-09-27:
@arichardson : Can we close this ticket?

arichardson, 2024-09-27:
Let's close this and keep it as is.

closed 2024-09-27 by arichardson