riscv/riscv-cheri #110: Behaviour of JVTC in legacy mode

arichardson, 2024-02-12:
Splitting out the remaning open issue from #43:
Currently JVTC resets to infinity. It would be nice to avoid yet another initial infinity registers.
Comments in issue #43 had a few suggestions of how to deal with this:

https://github.com/riscv/riscv-cheri/issues/43#issuecomment-1936807460:
> We could move away from "legacy writes to CLEN-bit CSRs preserve the metadata" and towards "legacy writes to CLEN-bit CSRs copy metadata from pcc". Then the jvtc metadata would be initialized automatically when legacy code initializes the jvtc address.

Or https://github.com/riscv/riscv-cheri/issues/43#issuecomment-1936170692:
> `table_access_ptr = setCapAddr(DDC, jvtc.address+index*XLEN/8)`

The first suggestion is quite appealing since it could potentially allow relaxing this restriction for further registers, but I need to think through the implications of this behavioural change a bit before being confident that it doesn't introduce any new issues. It somewhat restricts the ability to have separate ranges for these CSR for a constrained legacy mode program but I'm not sure that matters too much.

tariqkurd-repo, 2024-02-13:
ok - so is proposal 1 that every write to JVTC will automatically set the metadata from the PCC in legacy mode but allow full programming in capability mode?

I can understand removing the source of infinity, as it's an obvious security hole having it there.

As you said @arichardson it restricts the ability of controlling the range in legacy mode, but personally I think it's a sensible suggestion.

Capability mode retains everything it has now (with JVTC resetting to tag=0) and it remains compatible with the existing Zcmt spec as X permission is required.

arichardson, 2024-02-14:
@jrtc27 what do you think is the most sensible approach here?

If I read those issues correctly, the X requirement was just for memory fencing/page table reasons? Since it really is just a data fetch for an indirect jump I'd prefer we copy DDC metadata when writing jvtc over PCC, but I guess PCC could also work since it's supposed to be read-only data.

jrtc27, 2024-02-14:
I don't personally like this magic copying of metadata when you perform an integer write as a special case for this CSR. Logically it is an integer address in integer mode through which you are performing a load, and therefore it should be constrained by DDC at cm.j(al)t time.

tariqkurd-repo, 2024-02-23:
The current proposal is

1. reset to tag=0, otherwise undefined
2. in cap mode use JVTC to authorise the fetch from the table
3. in legacy mode use PCC metadata to authorise the fetch from the table

closed 2024-04-05 by tariqkurd-repo