riscv/riscv-cheri #134: Capabilities with non-zero reserved bits

jasonyu1996, 2024-02-27:
Hi!

In Section 2.1.7: 

    Reserved bits available for future extensions to Zcheri_purecap.
    Reserved bits must be 0 in valid capabilities

It's not clear to me what this means. If a future extension to Zcheri_purecap uses the reserved bits, those capabilities are not valid capabilities anymore? Or does this mean that the specifications in some parts of Zcheri_purecap only apply when the reserved bits are 0? If so, which parts?

andresag01, 2024-02-27:
A key aim of the CHERI specification is to leave sufficient bits in the capability encoding for future extensions. You can see how this works if you compare the capability encodings in Zcheri_purecap and Zcheri_mode:

* Zcheri_purecap is only for purecap machines and there is no "mode" bit
* Zcheri_mode supports the Legacy mode (formerly Integer in CHERIv9) and does require the "mode" bit in the capability encoding.

Therefore, Zcheri_mode has 1 less reserved bit than Zcheri_purecap. If your implementation supports Zcheri_mode then there is a valid set of operations by which you can legally obtain capabilities with that bit set to either 1 or 0. However, if your implementation supports Zcheri_purecap only, then there are no valid operations for you to legally obtain a valid (i.e. tag set) cap with that reserved bit set to anything other than 0.

This applies to future extensions on top of Zcheri_purecap. It also means that you cannot take an arbitrary program from Zcheri_mode, for example, and run it on a machine that just supports Zcheri_purecap. Although Zcheri_mode is backwards-compatible, so you can take a Zcheri_purecap program and run it on Zcheri_mode.

jasonyu1996, 2024-02-27:
> A key aim of the CHERI specification is to leave sufficient bits in the capability encoding for future extensions. You can see how this works if you compare the capability encodings in [...]

Thanks for the reply! I'm interested in learning about how an extension to Zcheri_purecap, not just the existing standard ones but also potential future custom ones, are supposed to work.

Your answer and the draft seem to suggest these (please correct me if I'm wrong):

- Zcheri_purecap only specifies what happens when the reserved bits is zero. Extensions to it specify what happens when the reserved bits take some other values.
- Extensions can override some part of Zcheri_purecap. In this case, at least the definition of a "valid capability" has changed.

I have a few more questions:

- If Zcheri_purecap only concerns capabilities with reserved bits equal to zero, doesn't an extension need to _explicitly_ specify which part of Zcheri_purecap also apply for "extended capabilities" that have non-zero reserved bits? For example, I suppose with Zcheri_purecap only, `CADD` on a capability with reserved bits = `0x1` is undefined, and an extension would need to specify what happens there, even if it simply wants to reuse the semantics for when reserved bits is zero.
- Is backward compatibility a requirement? Is it required that even with an extension, as long as all capabilities used have reserved bits equal to zero and all instructions use are those defined in Zcheri_purecap, the behaviour should be identical to when only Zcheri_purecap is available?

Thanks!

andresag01, 2024-02-27:
> Zcheri_purecap only specifies what happens when the reserved bits is zero.

The key to remember is that Zcheri_purecap does not make use of the reserved bits, so there is no valid set of instructions such that you will end up with a reserved bit set to something other than 0.

> Extensions to it specify what happens when the reserved bits take some other values.

Yes, if you want to propose an extension that allocates a reserved bit in Zcheri_purecap for something then you need to indicate what's that bit is going to do

> Is backward compatibility a requirement?

That's a good question. Ideally your extension would have some measure of compatibility with the base stuff, otherwise its like proposing an extension to RV32I, for example, that if implemented would not let you run programs that are simply RV32I. It would probably be difficult to find support for such an extension in the RISC-V community, hence why a lot of effort has been invested in providing extensions like Zcheri_legacy and Zcheri_mode that enable compatibility with RISC-V.

sorear, 2024-02-27:
Your "backward compatibility" requirement is failed by Zcheri_mode: `jalr` to an executable capability with all reserved bits clear executes Capability mode instructions in Zcheri_purecap, but Legacy mode instructions in Zcheri_mode. This is #34.

There are likely to be new permissions in the future, which will change the Zcheri_purecap initial value of `pcc` and `dinfc`, and some of which (Global most notably) will conspicuously change the behavior of Zcheri_purecap instructions when clear. This is essentially #71.

andresag01, 2024-02-27:
@sorear: Apologies if I gave you the wrong impression! I never said it was a "requirement"; I said that "ideally" there is "some measure of compatibility" which in no way implies that there is a rule. In fact, I would expect that each extension will need to be reviewed and compatibility assessed on a case-by-case basis.

arichardson, 2024-02-27:
The reserved bits only need to be zero if the core does not implement any extensions that assign meaning to them. I see this as similar to CSRs where bits are later defined to have a meaning and older cores that don't implement them have them as read-only zero.

The only time these reserved bits matter is when using CBLD where it should reject any bitpattern that has bits set that the current CPU does not assign a meaning to. This ensures that it's impossible to ever encounter a capability that has reserved bits set. The only exception would be a system with multiple CHERI cores that use different extensions and therefore incompatible capability formats. If one of them sees a capability with reserved bits set it must treat it as invalid even if it has the tag set in memory. Although to be honest I think such a system is going to have a lot of issues beyond the reserved bits if there is any shared memory and should not be supported.

Maybe the spec needs to be more clear that the number of reserved bits depends on which of the extensions are supported?

tariqkurd-repo, 2024-04-04:
What the spec is intended to say is that it is impossible to have a tagged capability with any of the reserved bits set, or any of the reserved permissions combinations for RV32 (for https://github.com/riscv/riscv-cheri/pull/155).

Future extensions will allow such capabilities, and the definition of how they behave will have to be documented.

For example the proposed local/global mechanism will do exactly this. It will allow currently reserved states to be exist in tagged capabilities, and the effect on all instructions and CHERI behaviour will need to be specified. Ideally those reserved states will be uniquely allocated to local/global, and there'll be no future conflict with other extensions, although this is possible if we run out of space in the capability format (only likely for RV32).

So we're really treating it like the RISC-V encoding space, and filling in reserved opcodes as we go.

Is that helpful?

andresag01, 2024-04-11:
Is there anything further to do on this ticket? Can we close it?

andresag01, 2024-04-22:
It looks like the discussion concluded, so I will close this ticket. We can always reopen it (or create a new one!) if there are further questions.

closed 2024-04-22 by andresag01