riscv/riscv-cheri #139: Specify how invalid address conversion is handled for jumps.

Timmmm, 2024-03-08:
For JAL in legacy mode it says:

> the target address which is written to the [pcc](https://riscv.github.io/riscv-cheri/#pcc)'s address field

(`the` should probably be removed btw)

It says a very similar thing for conditional branches.

It doesn't really specify how invalid address handling should be applied here. Suggestion from Jamie was that the (non-capability) target address undergoes invalid address conversion. PCC is not affected if invalid address conversion changes the target address.

PRugg-Cap, 2024-03-29:
@andresag01 and I had some discussion about this a while back: I think the plan was for the architecture to specify that the bounds-check happens pre-conversion. The conversion in the architecture (with representability check) should only happen on the subsequent fetch when the translation has failed and working out what to put in epc(c).

Happy to PR if we agree this is a sensible approach?

andresag01, 2024-04-02:
@Timmmm : You are right, the spec is missing a description for pcc handling -- we only have a description for CSRs

@PRugg-Cap: I seemed to remember that it would work like this (e.g.) in the case of JAL:
```
// Get an XLEN target
targetAddr = pcc.address + imm
// Drop unused bits, then decompress to extend back to XLEN to see whether we lost precision due to invalid address
targetAddr = COMPRESS(DECOMPRESS(targetAddr))
// Check bounds as usual
if inCapBounds(pcc, targetAddr)
    pcc = setCapAddr(pcc, targetAddr)
else
    EXCEPTION
```

PRugg-Cap, 2024-04-02:
Hmm, I don't think that was ever my idea of how it would work: it would be nice that this didn't pollute the architecture more than it has to. It really should just affect exception reporting, so why would you want to do the transformation before the bounds check?

PRugg-Cap, 2024-04-02:
I might be misremembering though: I'll have a think

andresag01, 2024-04-02:
> It really should just affect exception reporting, so why would you want to do the transformation before the bounds check?

You are suggesting that we just assign the target address to pcc without checking bounds or representability. If so, then we run the risk of manufacturing a capability that breaks the encoding and is still tagged. I do not think we can just wait for the next instruction to check the PCC because the encoding would have been broken by then (assuming that PCC.address only has sufficient bits to hold the valid addresses)

jrtc27, 2024-04-02:
I’m of the same view as Peter, I don’t think you should be mucking with the address. But also if you fail the bounds check then you don’t perform the jump so there is no invalid address to write anywhere, the faulting PC(C) is that if the jump itself.

PRugg-Cap, 2024-04-02:
@andresag01 I'd consider it more as that you jump to the unconverted address. The architecture only tells you that epc(c) doesn't need to have the redundant bits. The microarchitecture still doesn't have to store those bits: the jump goes through then the microarchitecture throws the bits away while noting to itself that the CHERI fetch checks should pass, then the illegal fetch translation happens as it always did.

PRugg-Cap, 2024-04-02:
@andresag01 I'm not suggesting you put some hacky bit of state directly forwarded between the instructions or anything. I'm saying that you "compress" a PC in the microarchitecture to include a bit that says it's been through all the CHERI checks. If you take an interrupt in the meantime, then the architecture is allowed to compress the PCC and it's okay for that bit to get thrown away along with the redundant high address bits.

tariqkurd-repo, 2024-04-03:
@arichardson any thoughts?

PRugg-Cap, 2024-04-03:
@tariqkurd-repo This will have some odd implications though, e.g. I can check if a cap is in-bounds, but then get a bounds exception when I jump to it. It sounds the same as saying that we should be allowed to throw an instruction misaligned fault on the jump because the implementation might legalise it to an unaligned address.

I see what you're saying about needing a representability check when compressing the cap into EPCC, though, and it would be sad to insist on that without a good reason. I wonder if the spec should just say the tag gets cleared if the fetch address was invalid, i.e. if the legalisation isn't a nop?

andresag01, 2024-04-03:
> This will have some odd implications though, e.g. I can check if a cap is in-bounds, but then get a bounds exception when I jump to it.

The argument is that the legalization for the invalid address is an architecturally-defined behaviour for all CSRs that hold addresses. Therefore, manually checking the bounds on an XLEN register is naturally going to give a different result from doing the analogous computation on a trimmed down register like ddc, pcc, mtvecc, stvecc, etc.

I agree that its quirky, but it needs to work somehow. At the end of the day, this is an optimization, so spending lots of hardware on additional representability checks for branches/jumps instead to save flops for PC bits would be unfortunate.

> I wonder if the spec should just say the tag gets cleared if the fetch address was invalid, i.e. if the legalisation isn't a nop?

We previously considered just clearing the tag when an invalid address occurs (not just in branches/jumps), but discarded the idea because its almost like saying that invalid address exceptions don't exist and we instead get a CHERI tag violation which is not very helpful.

PRugg-Cap, 2024-04-03:
> The argument is that the legalization for the invalid address is an architecturally-defined behaviour for all CSRs that hold addresses. Therefore, manually checking the bounds on an XLEN register is [...]

The issue I have is that something that's supposed to be a slight change to exception reporting has effectively turned into another layer of translation. If we're defining that bounds checks happen on the post-conversion address, why shouldn't that happen to data memory accesses as well? What if the illegal address legalisation isn't continuous, i.e. some of the access is in-bounds and some is not? I'm worried also that an access that is supposed to be out-of-bounds could become in-bounds after the conversion, rather than just the other way around.

> I agree that its quirky, but it needs to work somehow. At the end of the day, this is an optimization, so spending lots of hardware on additional representability checks for branches/jumps instead to save flops for PC bits would be unfortunate.

I absolutely agree it has to work. Just trying to think about ways to make it work cheaply without as many quirks.

> We previously considered just clearing the tag when an invalid address occurs (not just in branches/jumps), but discarded the idea because its almost like saying that invalid address exceptions don't exist and we instead get a CHERI tag violation which is not very helpful.

Okay, I was just proposing clearing the tag when it got put into EPCC, but I realise that has the problem that a completely CHERI-oblivious software stack could take a tag exception when xRETting after it's been unlucky enough to take an interrupt just after doing one of these dodgy jumps but before the subsequent fetch check. I'll think a little bit more about whether it's possible to avoid a representability check per branch unit, but otherwise I agree with @tariqkurd-repo's summary of what this would mean pragmatically.

tariqkurd-repo, 2024-04-04:
I *think* there a simple solution here.

1. for legacy mode compatibility we must take the instruction access fault, not a tag fault in these cases, _if_ the cap is infinite. if it isn't then we _can_ clear the tag as we're using a CHERI-aware OS.
2. clear the tag on invalid addresses detection if the cap _isn't_ infinite

Therefore if the cap isn't infinite we get a tag violation, if it _is_ infinite we get the correct RISC-V behaviour of the instruction access fault.

How does that sound?

I think this is also applicable to load/store. No-one can load/store/fetch from invalid addresses - the only question is which type of fault is taken.

No additional representability checking logic required.

Timmmm, 2024-04-04:
Bit of a shame that the other CSR writes (e.g. `mtvecc`) use a slightly different algorithm to writing to PCC though.

Maybe a stupid idea but could you use the normal invalid address conversion algorithm (i.e. clear the tag if it is invalid) when writing PCC but then also say that instruction access faults due to invalid addresses are simply higher priority than CHERI tag checks?

tariqkurd-repo, 2024-04-04:
The CHERI fault must come first in all cases, so to maintain legacy compatibility we need the tag to be set for invalid addresses, which is easy if the metadata is infinity.

>2. If conversion was required, then A’s tag is cleared if A is sealed or if the new address is not representable — this is equivalent to the semantics of SCADDR

from `3.13.1. Invalid Address Handling`

If we changed this from `new address is not representable` to `address changed and the metadata is not infinity` then it all fits together. How about that? @PeterRugg ?

PRugg-Cap, 2024-04-04:
@tariqkurd-repo Yes, I was about to suggest something similar. Either define that they're cleared when the bounds are not almighty or make it part of the implementation-defined legalisation that the tag is allowed to be cleared so long as it's never cleared when the bounds are almighty, but is cleared at least in the case where the bounds interpretation would change.

Even if the tag will be clear when the PCC gets written to EPCC, the architecture currently still says that conversion only happens on writing to EPCC, so the CHERI fetch checks immediately after the jump/branch would still have to pass. I think that's handled by the "flag" suggested above: basically the microarchitecture is storing the whole PCC for that stretch but "compressing" it down to a single bit that's enough to say it should pass the checks. Does that make sense, or should we try to simplify things a bit?

@Timmmm It is tempting, but I don't think it avoids the representability check: you still need to ensure the EPCC is tagged if the bounds are almighty, but cleared if the bounds interpretation would change, otherwise you're just delaying the unexpected CHERI fault until the next xRET.

PRugg-Cap, 2024-04-04:
> If we changed this from new address is not representable to address changed and the metadata is not infinity then it all fits together.

I think that may be okay: I'll just have a look through to make sure there aren't other cases where we'd regret this.

andresag01, 2024-04-04:
Conclusion from today's conversation:
1. Do invalid address conversion _before_ any CHERI checks are performed
2. The branch/jump throws a tag violation if, after conversion, the address is invalid and PCC is NOT Infinite.
3. The branch/jump throws  instruction access fault if, after conversion, the address is invalid and PCC is Infinite. This is to ensure that Legacy + Infinite is compatible with standard RISC-V.

@PRugg-Cap @tariqkurd-repo @arichardson @jrtc27 : Do you agree with this behaviour?

Timmmm, 2024-04-05:
And just to be clear, if you are in legacy mode and write a capability with an invalid address and infinite bounds to `mtvec`, would it clear the tag?

Would it make sense to just change the invalid address conversion procedure so that all invalid address conversions special-case infinity. I.e. instead of 

```
function setCapAddrChecked(c : Capability, addr : CapAddrBits) -> Capability =
  let (representable, newCap) = setCapAddr(c, addr) in
  clearTagIf(newCap, not(representable) | isCapSealed(c))

function convertInvalidAddrCap(cap : Capability) -> Capability = {
  let reducedAddr = convertInvalidAddr(cap.address);
  if reducedAddr != cap.address then setCapAddrChecked(cap, reducedAddr) else cap
}
```

do

```
function convertInvalidAddrCap(cap : Capability) -> Capability = {
  let reducedAddr = convertInvalidAddr(cap.address);
  if reducedAddr != cap.address & not(hasInfiniteBounds(cap)) then setCapAddrChecked(cap, reducedAddr) else cap
}
```

Also `hasInfiniteBounds()` needs to be specified since IIUI there's more than one possible cap that covers all the address space. Do we mean:

```
function hasInfiniteBounds(cap : Capability) -> bool =
  cap.E == infinite_cap.E & cap.B == infinite_cap.B & cap.T == infinite_cap.T & cap.internal_exponent == infinite_cap.internal_exponent
```

or this?

```
function hasInfiniteBounds(cap : Capability) -> bool = {
  let (base, top) = getCapBounds(cap);
  base == 0 & top >= 2^xlen
}
```

PRugg-Cap, 2024-04-05:
> And just to be clear, if you are in legacy mode and write a capability with an invalid address and infinite bounds to mtvec, would it clear the tag?

It would not clear the tag. Otherwise software can end up getting a tag exception without ever touching capability features.

> Would it make sense to just change the invalid address conversion procedure so that all invalid address conversions special-case infinity.

I think so! We should check, but I don't think there's any reason not to make this behaviour universal for invalid address conversions. It should be a lot cheaper. I think the proposal is slightly different to the sail snippet though: it should be more like:

```
function convertInvalidAddrCap(cap : Capability) -> Capability = {
  let reducedAddr = convertInvalidAddr(cap.address);
  clearTagIf(reducedAddr != cap.address & not(hasInfiniteBounds(cap)), cap)
}
```
I.e. we're saying the representability check doesn't happen at all and you just blindly clear the tag if you change the address.

> Also hasInfiniteBounds() needs to be specified since IIUI there's more than one possible cap that covers all the address space.

Sort of. There's only one set of bounds that should ever be tagged that covers the whole address space, so you shouldn't be able to tell the difference between the two checks you proposed anywhere. For clarity, I'd suggest we go with the `base == 0 & top >= 2^xlen` definition.

tariqkurd-repo, 2024-04-05:
I think we're in agreement - we'd apply the same behaviour _everywhere_. So if you write an invlaid address to metvec and the metadata isn't infinite then you clear the tag

PRugg-Cap, 2024-04-12:
Agreed that this seems a good way to go. Minor nit, just to check I understand correctly:

> The branch/jump throws a new CHERI invalid address fault if, after conversion, the address is invalid and PCC is NOT Infinite.

The "after conversion" is redundant here? Illegal addresses have to convert to illegal addresses?

PR #188 (2024-04-12): Change CSR/jump/branch invalid address handling
* Unify the invalid address handling for both CSR and control transfer instructions
* The new invalid address handling clears the tag if the capability being written is not Infinity. Also, invalid address conversion is now performed before anything else

Fixes #139