riscv/riscv-cheri #143: load/store/atomic opcode collisions

sorear, 2024-03-11:
We're not the only group to have had the idea to use 2*XLEN bit operations. Zilsd uses LQ/SQ on RV64 and LD/SD on RV32 and will collide with LC/SC on Zcheri_purecap. Zacas defines AMOCAS.D on RV32 and AMOCAS.Q on RV64 for double-register operations, which will collide with AMOCAS.C if it is defined in the obvious way.

Zacas is the larger problem since it's part of the current draft for the RVA23 profile (despite being unusable by compilers without a double-XLEN load extension that isn't defined yet).

I would strongly prefer to keep the width encoding the same between all atomic operations, i.e. AMOCAS.C and AMOSWAP.C use the same funct3.

Do we want to forbid AMOCAS.Q in RV64 Zcheri_legacy, sacrificing RVA23 compatibility?

Do we want to forbid AMOCAS.Q in RV64 Capability mode and AMOCAS.C in RV64 Legacy mode, increasing the divergence between Capability and Legacy mode? If so, is this an argument in favor of disallowing all capability loads and stores in Legacy mode?

Do we want to change the funct3 used by AMOCAS.C (RV64), and by extension AMOSWAP.C, LR.C, and SC.C, to something other than 4? If so, should LC and SC also move?

tariqkurd-repo, 2024-03-12:
> My current feeling is that when we finally get to ARC review all the
> encodings are going to change anyway, so it's not obvious that it's worth
> changing anything now.
>
> That's why this isn't a PR. Asking the ARC to deal with this is fine. Not
> even documenting the issue makes their lives harder for no reason, I feel.

When I spoke to Andrew at the RISC-V summit his opinion was that reusing
the LQ/SQ (proposed) encodings made sense, so it shouldn't be a surprise.

tariqkurd-repo, 2024-03-13:
having said they anything that knowingly prevents RVA23 from being implemented as legacy mode is a real problem - so thanks for pointing this out

tariqkurd-repo, 2024-09-30:
We've moved encodings to avoid Zilsd now, and will hopefully get to ARC review to allocate encodings soon

closed 2024-09-30 by tariqkurd-repo