riscv/riscv-cheri #149: Top needs to be 66 bits

PRugg-Cap, 2024-03-21:
(Using RV64 for clarity, but obviously 34 bits in RV32 etc.)

There's currently a case where base > top, reminding me that Cambridge's "underivable" branch needed to move to 66-bit arithmetic for top: https://github.com/CTSRD-CHERI/sail-cheri-riscv/tree/underivable.

This is to address the fairly specific issue where a cap has:
 - E = max_E (52), meaning the top two bits of B and the (reconstituted) top two bits of T "poke out" over the top of the 64-bit address space. (The only tagged cap that can exist with E=52 has B=T=0 and is the almighty cap).
 - B is zero in the top two bits (otherwise it would be malformed) and non-zero in the remaining bits.
 - T is less than B, making the top two bits of T 00 + 1 (implied MSB) + 1 (T < B), and setting bit 65.

If the 66-bit arithmetic is too painful for hardware, we may also be able to fix this in the malformed check, but that would be the first case (I think) where you can become malformed without twiddling the bits directly using `CSetHigh` or round-tripping through memory. The above capability can be constructed (untagged) simply by setting the address of a cap to -1 and requesting length -1. Changing it in the malformed check would also make the malformed check dependent on the full mantissa, whereas up to now it only depended on the exponent and top two bits of B.

`CBuildCap` currently has a check that `requestedBase <= requestedTop` (https://github.com/CTSRD-CHERI/sail-cheri-riscv/blob/6e3613a2c46fb809e526b55c5c72acb041194ab8/src/cheri_insts.sail#L803), which should be able to go away with this fix. We'd want to prove this first though!

tariqkurd-repo, 2024-04-02:
>and requesting length -1

is this possible? the `scbnds` instruction page doesn't say anything about the sign of `rs2` - shouldn't it be an unsigned operand?

tariqkurd-repo, 2024-04-02:
Can we special case E=52 to avoid all this? Are there's only one valid cap with that exponent?

PRugg-Cap, 2024-04-02:
By "special case" I guess you mean "make malformed"? The malformed check would depend on the the mantissa bits then, but I guess only on whether they're the specific allowed value rather than on a subtract.

PRugg-Cap, 2024-04-02:
It still means you can go malformed fairly easily with common instructions, but I doubt that's a real problem.

tariqkurd-repo, 2024-04-02:
>The only tagged cap that can exist with E=52 has B=T=0 and is the almighty cap

so can we say that if E=52 AND (T!=0 OR B!=0) then the cap is malformed? Would that avoid the mantisssa check?

PRugg-Cap, 2024-04-02:
As in, that is a mantissa check, but yes, it avoids needing to subtract them at least.

tariqkurd-repo, 2024-04-02:
I was confusing mantissa with the address field (my brain going back to floating point terminology).
E=52, T==0, B==0 are cheap to compute, so I'm happy with that feeding into the malformed check.
Time for a PR?

PRugg-Cap, 2024-04-02:
Okay, I'll do my best to make sure it solves the problem then PR it in.

closed 2024-04-10 by arichardson