riscv/riscv-cheri #154: Should PCC be a CSR in Zcheri-legay?

veselypeta, 2024-03-22:
https://github.com/riscv/riscv-cheri/pull/113 Removed the PCC CSR register. This was previously used to implement converting pointers to capabilities where it's bounds are derived from the pcc. If `modesw` is availble you can read the pcc using `auipc`. Would it be better to keep the PCC CSR register for this purpose?

PRugg-Cap, 2024-03-29:
Hmm, agreed, this sounds like a problem: the current spec prevents a legacy-mode process from getting anything PCC-relative without Zcheri_mode. This was mentioned in https://github.com/riscv/riscv-cheri/issues/90: we no longer need the PCC CSR to facilitate a mode switch, but I guess we didn't consider that a legacy process would want access to PCC-relative data.
I suppose we should consider whether there's actually a use-case for this? If the region used isn't also covered by DDC then the process has no hope of actually dereferencing the derived capability, but could still hand it out to other cap-mode processes to use. Sounds a bit odd, but I wouldn't want to say for sure that it's useless/useful.

andresag01, 2024-04-03:
There is currently a proposal to merge Zcheri_legacy and Zcheri_mode into a single extension, so the `modesw` would always be available and the problem described here would be resolved. Lets keep this ticket open until #161 is merged.

andresag01, 2024-04-03:
#161 is due to get merged soon, so this is resolved.

closed 2024-04-03 by andresag01