riscv/riscv-cheri #156: Sdtrig exception priorities

tomaird, 2024-03-25:
The `Sdtrig` specification (RISC-V Debug Spec) describes a list of synchronous exception priorities. 

The CHERI spec makes no reference to how the new CHERI exceptions fit into this priority list.

e.g. We have a `mcontrol6` trigger configured to match on load addresses, with timing "before", and `action` set to raise a breakpoint exception. Then we do a load with an `auth_cap` without load permissions (or no tag etc.). Does this fire the trigger and raise a breakpoint exception, or the CHERI exception?

I'm not sure if there are any other interactions between CHERI and Sdtrig that need considering too.

tariqkurd-repo, 2024-03-26:
Breakpoints must always be higher priority otherwise debug wouldn't be possible.
This should be in the doc...... that was the intention.....

andresag01, 2024-04-02:
@tomaird is right in that we have the exception priorities but these do not include the Debug exception priorities. We just need to add the equivalent table in the debug integration chapter of the CHERI spec -- Its essentially the same table we already have with a few more rows added in