riscv/riscv-cheri #158: Specify reset state of general-purpose register file

PRugg-Cap, 2024-03-27:
In the introduction we claim capabilities are made available in the general-purpose register file on reset: https://github.com/riscv/riscv-cheri/blob/22a47dcd1c1549a95fbecfd5fef019e7a6ba114c/src/introduction.adoc?plain=1#L69-L70

I don't think there's any other mention of reset state of general-purpose registers currently. We need to constrain them at least enough to ensure there's nothing malformed and tagged. It would be tempting to say that no GPRs are tagged on reset (auipcc or csrr ddc to bootstrap), and note that then we'd need to change that intro text.

arichardson, 2024-03-27:
Agreed, we only expose known values of capabilities via CSRs.

It would also be nice to say that registers are untagged on reset but I'm not sure this is strictly required.

The startup code already has access to an almighty PCC, so stale capabilities are not a problem in terms of privilege escalation. Any startup code should zero registers anyway before handing over control to the next boot stage.

The one thing that does sound like a problem is reserved patterns if the reset state is completely arbitrary rather than just stale values from a warm reset.

andresag01, 2024-04-02:
I agree that this needs to be tidied up. We should make it such that special capabilities are in CSRS and PCC while the other general purpose registers must have their tag cleared at reset. This prevents having arbitrary values (some possibly breaking the CHERI Concentrate encoding) around.

PR #170 (2024-04-03): Indicate that GPRs have tag cleared on reset
Fixes #158
commit 3bb7d5f8 (2024-04-03): Indicate that GPRs have tag cleared on reset (#170)

Fixes #158
closed 2024-04-03 by arichardson