riscv/riscv-cheri #159: Zba: should add.uw be mode-dependent?

PRugg-Cap, 2024-03-28:
The Zba extension is for bit manipulations related to address generation. As I understand it, it consists of:

 - sh{n}add{.uw} where n is {1,2,3} to take an index, shift it, and add it to a pointer. With uw the index is a 32-bit unsigned value.
 - add.uw to take a 32-bit unsigned index, not shift it, and add it to a pointer.
 - slli.uw to shift an index: this is to use with add (non uw!) in cases where you want to shift the index by more than 3.

We have currently made all the sh{n}add{.uw}s mode dependent, which seems very sensible, and also added sh4add{.uw}, presumably because we imagine a common case where indices are into tables of caps. It seems sh4add{.uw} could be made RV64-only since shift-by-3 is sufficient for 8-byte caps: I doubt it's worth the hassle, but sh4add may imply an extra layer of muxes...

add.uw seems to have been missed. This seems to be filling the gap of sh0add.uw, so I don't see why it should be treated differently. I guess the non uw sh0add is just add, and we do treat that one a little differently in that we have a separate instruction for cadd!

slli.uw should be left un-capified because it's for manipulating the index before it gets combined with the pointer, if I understand correctly.

Hope that makes sense: happy to file a PR to make add.uw mode-dependent if people confirm I haven't missed a good reason it should be left out.

arichardson, 2024-03-28:
I'm not quite sure how useful add.uw is but for consistency I agree that it should be treated the same way. Also agreed that shadd4 should depend on zba.

PRugg-Cap, 2024-03-28:
https://github.com/riscv/riscv-cheri/pull/160 fixes these and also changes some encodings: I'm fairly sure this is what was intended.

The only remaining question I have is whether sh4add{.uw} should be RV64 only.

closed 2024-04-02 by tariqkurd-repo