riscv/riscv-cheri #163: Allowing delayed address bound checking for jumps and branches

kliuMsft, 2024-04-02:
One thing we have learned from the cheriot-ibex design is that it helps quite a bit to allow delayed address bound checking for jumps and branches for MCU-type designs. Basically, to allow the target instruction to be fetched but compare the corresponding PC against the PCC bounds before it can be executed.

The main benefits are,
1. Removing the comparators from the instruction fetch timing paths (which is especially good for MCU's with smaller number of pipeline stages and often generates fetch requests combinatorically to reduce fetch latency. In contrast, by delaying the checking, the comparison has at least 1 extra cycle to finish, which likely would help reducing power consumption as well.

2. Reducing design complexity and cost of adoption cost (if cheri support is added to an existing microarchitecture). Checking address bounds at fetching time covers all cases (jal/jalr, conditional branches, traps/syscalls). And there is no need to modify the existing logic for branch instructions, which saves significant design and verification effort.

It's true this is essentially a trade-off between design/timing and debuggability and recoverability, as delayed checking makes harder to pinpoint the violating jump/branches and recover from them. However that could be a solvable problem with a bit of additional debug/trace information.

@rmn30 @nwf-msr @davidchisnall for more comments.

jrtc27, 2024-04-02:
This looks to be rehashing https://github.com/riscv/riscv-cheri/issues/6. It's not just a debugging issue, it is a functional issue for compartmentalised address spaces.

jrtc27, 2024-04-02:
(Closed as duplicate)

closed 2024-04-02 by jrtc27

davidchisnall, 2024-04-02:
Please can you elaborate on the functional issues for compartmentalisation? In CHERIoT, we’ve built a rich compartmentalisation model with the checks in the locations @kliuMsft describes and not had any problems.