riscv/riscv-cheri #186: cbld description does not match code snippet

tomaird, 2024-04-08:
The description for `cbld` doesn’t match the code snippet.

The description says “copy `cs2` to `cd` and set the tag to 1 if `cs1.tag` is set” but it makes no mention of **clearing** the tag if `cs1.tag` is **not** set.

[image]

The code snippet says that you clear the tag of `cd` if `cs1.tag` is not set. 

[image]

closed 2024-04-09 by andresag01

PRugg-Cap, 2024-04-12:
Hmm, just spotted this: it mostly doesn't matter, but it can help to leave the tag set when you want to try to buildcap with a chain of possible authorising caps, so we may want the "leave the tag set" behaviour. @jrtc27 has done some work on this and may have thoughts?

arichardson, 2024-04-12:
> Hmm, just spotted this: it mostly doesn't matter, but it can help to leave the tag set when you want to try to buildcap with a chain of possible authorising caps, so we may want the "leave the tag set" behaviour. @jrtc27 has done some work on this and may have thoughts?

Are you suggesting we just treat this as a move when the input.tag == 1 && auth.tag == 1?

PR #189 (2024-04-09): Clarify that cbld cd.tag=0 if checks fail
Fixes #186
commit d55a2b84 (2024-04-09): Clarify that cbld cd.tag=0 if checks fail (#189)

Fixes #186