riscv/riscv-cheri #198: Reset value for dscratch{0,1}[c] CSRs

AJoannou-Cap, 2024-04-12:
I was browsing the spec for [dscratch0](https://riscv.github.io/riscv-cheri/#dscratch0) and [dscratch0c](https://riscv.github.io/riscv-cheri/#dscratch0c) and was not sure if I wrapped my head around the text correctly...

As I understand it, the `dscratch0` section specifies that it is extended to `dscratch0c` with reset value tag=0, metadata=unspecified, address=unspecified. The section on `dscratch0c` specifies its reset value to be the `NULL` cap ([whose fields are all specified to be 0](https://riscv.github.io/riscv-cheri/#section_null_cap)).

I presume the intention is for the `NULL` cap to be in `dscratch0c`, meaning that the reset value of `dscratch0` should be 0... ?
If this is correct, I am happy to propose a PR to fix this.

andresag01, 2024-04-12:
Thanks for pointing this out!

The dscratch0c/dscratch1c are both supposed to have tag=0 metadata=UNSPECIFIED address=UNSPECIFIED. This is written elsewhere in the document, but we missed the text you rightly pointed out in the sections describing those CSRs.

AJoannou-Cap, 2024-04-15:
I just opened #200

andresag01, 2024-04-16:
Thank you for your contribution. #200 has been merged, so this ticket can now be closed.

closed 2024-04-16 by andresag01