riscv/riscv-cheri #199: Instruction for unsealing sentries with a superset

arichardson, 2024-04-12:
In ISAv9 CBuildCap was usable for unsealing sentries since it cleared the otype field. This is no longer the case, so unsealing a sentry with a superset is somewhat awkward. I believe we need the following sequence:

```
gctag t0, <sentry>
bnez t0, .Lfail
gchi t0, <sentry>
and t0, t0, ~SEALED_BIT
schi <sentry>, <sentry>, t0
cbld <unsealed>, <superset>, <sentry>
// Start using result if tag is set
gctag t0, <unsealed>
bnez t0, .Lfail
```

It would be much nicer to have a variant of cbld that did the otype clearing so we can use
```
unsentry <unsealed>, <superset>, <sentry>
gctag t0, <unsealed>
bnez t0, .Lfail
```

The checks this new instruction would need to do are:
- Is tag set?
- Is otype == sentry
- If so, clear otype and use remaining cbld logic
- else return input with tag=0

Almost all the logic is shared with cbld so I don't think this would add much complexity, only need to reserve one 3 operand instruction.

Naming is hard, could use something like `unsentry` or `ssu` (superset unseal) depending on how we think it will interact with a future otypes extension.

jrtc27, 2024-04-12:
That’s not true for ISAv9; our CBuildCap creates sentries. Morello has the behaviour you’re talking about.

arichardson, 2024-04-12:
> That’s not true for ISAv9; our CBuildCap creates sentries. Morello has the behaviour you’re talking about.

I'm pretty sure that at some point it was possible to feed a tagged sentry to cbuildcap and receive a tagged unsealed cap. Probably changed when ccopytype was added? Will update the description.

jrtc27, 2025-05-21:
It's a straight up specification bug that it takes so many instructions to unseal, rendering sealing pretty useless with regards to performance. This needs to be fixed prior to ratification (we don't seem to have a label for that?).

davidchisnall, 2025-05-22:
Fast unsealing is not that important for sentries, it’s much more important for generic otypes (where you may need multiple unseals on entry to a function if it’s exposed across compartments). Generic otypes are not part of the standard proposed for ratification. Most uses of generic otypes do not have access to the superset capability of the sealed thing and so cannot use a buildcap-based unsealing mechanism (in a properly privilege-separated unwinder, the unwinder would *also* not have this access and so would not be able to use buildcap to unseal: feedback from MSRC on prior CHERI specs pointed out that stack unwinders are already a high-value target for attackers and being able to deprivilege them is an important benefit of CHERI).

I am not convinced that the sentry-only sealing model is sufficiently baked to be part of the base specification, perhaps it should be moved to a Ysentry extension?

PR #767 (2025-09-19): Split reconstruction and superset unsealing
This is an alternative to #760 as per discussion over there.  It introduces a notion of ambient sealing types, defines `YBLD` explicitly as sealing those types, and defines type `1` as ambient.  Separately, it introduces a `YSUNSEAL` instruction for explicit superset unsealing, which is a lot like `YBLD` but requires a tagged input.

There are two other commits here that are purely clerical: adding xrefs and changing the variable names, not their rendered values, of the sentry instruction extension.

If this is agreeable, I will rework #716 according [... cut]
closed 2025-09-19 by arichardson