riscv/riscv-cheri #218: Clarification on CRE's effect on CSR accesses

jamie-melling, 2024-04-25:
I think that section 5.7. Disable CHERI Registers needs more information regarding its effects on CSR accesses.  All that is mentioned is that:

>Executing CSR instructions accessing any capability wide CSR addresses (Section 3.6) cause illegal instruction exceptions

There is no mention of DDC, MTDC, STDC.
Furthermore, without mention of those registers it leaves this question: if mseccfg.CRE is 0, does this mean any accesses to the XLEN address field of mtvec, mepc, mscratch cause illegal instruction exceptions?

The statement in the spec does not distinguish between RISC-V CSR extended by Zpurecap and the new CHERI CSRs that Zcherilegacy adds. (Maybe there is no difference, but you cannot tell)

andresag01, 2024-04-25:
> There is no mention of DDC, MTDC, STDC.

These are all CLEN CSRs, so they would would trap when CHERI is disabled.

> if mseccfg.CRE is 0, does this mean any accesses to the XLEN address field of mtvec, mepc, mscratch cause illegal instruction exceptions?

They would not trap. Recall that we are trying to maximize compatibility with non-CHERI RISC-V, so the regular XLEN-sized CSRs must remain accessible as usual

jamie-melling, 2024-04-25:
The mode is forced to legacy when CHERI is disabled, so you will never be able to do a CLEN operation (only XLEN) on the CSRs mentioned in 3.6, so I don't think it should refer to section 3.6 then.  It should instead just mention DDC, MTDC and STDC.

PRugg-Cap, 2024-04-25:
@jamie-melling Agreed: I don't think this was updated with the new "CLEN and XLEN CSRs alias and are mode-dependent behaviour". Therefore, any existing RISC-V CSR is safe to access, as mode is hardwired to legacy so you'll only be accessing the address, and we need to be compatible, as @andresag01 said.

The question is whether the extra capability CSRs without an existing RISC-V version can have their addresses accessed with CRE=0. It probably doesn't matter too much either way: if they've been subset (i.e. some code ran to restrict their bounds while CRE=1, then CRE was turned back to 0) then that could allow code to clear the tag by making them unrepresentable, but that goes for the ones with non-CHERI aliases as well anyway...

jrtc27, 2024-04-25:
They have to be inaccessible, otherwise that's a new side-channel that CHERI-unaware software leaves open if you run a CHERI-aware attacker on top of it

PRugg-Cap, 2024-04-26:
Oh yes, good point!

tariqkurd-repo, 2024-04-26:
yes - I'm expecting legacy mode to look like a normal RISC-V core (XLEN access to MTVEC etc.)
CRE=1 gives access to the new CSRs in legacy mode, but with XLEN access only
CRE=1 gives the ability to enter capability mode, and so CLEN access to MTVECC etc.

what has just sprung to mind is that we have XLEN access to DDC only in legacy mode.
which I think is implied by the text but maybe not explicitly stated (and MTDC etc.).

So maybe that part could be clearer but I don't think there are any actualy spec bugs here?

arichardson, 2024-05-03:
This should hopefully be addressed now?

closed 2024-05-07 by andresag01