riscv/riscv-cheri #220: Extension names don't follow the official convention

Timmmm, 2024-04-26:
Very minor but we have `Zcheripurecap`, `Zcheripte`, `Zcherilegacy`, `Zstid`, 

From the unprivileged spec:

> The first letter following the "Z" conventionally indicates the most closely related alphabetical extension category, IMAFDQCVH. For the "Zam" extension for misaligned atomics, for example, the letter "a" indicates the extension is related to the "A" standard extension.

So I think they should be `Zicheripurecap`, `Zistid`, etc.

tariqkurd-repo, 2024-04-26:
We know that `c` is wrong - but we're hoping to get an official letter allocated.
Whatever we call it will be changed again so I don't see a strong reason to change it now.

tariqkurd-repo, 2025-06-10:
duplicate of https://github.com/riscv/riscv-cheri/issues/535