riscv/riscv-cheri #229: Clarify whether nonsense permission combinations are banned in RV64

PRugg-Cap, 2024-04-30:
We currently carefully define the allowed permission encoding combinations in RV32, leaving some reserved. There's also a note describing how to treat the reserved combinations: https://github.com/riscv/riscv-cheri/blob/e3c20ffbfeebe4432a8f3994b31eed81374f297b/src/cap-description.adoc?plain=1#L184-L190

For RV64, we don't compress the permissions, so could in principle go without reserving any. However, we ban transitions with ACPERMS into the combinations deemed meaningless. I think the current spec is unclear as to whether these encodings should be treated as reserved if loaded in from memory or via schi. If they are not reserved, banning the transitions seems superfluous, so I assume they should be reserved and the above note should apply to RV64 as well?

tariqkurd-repo, 2024-05-02:
The reason not to use the same encoding scheme for RV64 was to allow more combinations, so if we ban them all requiring a spin of the silicon to allow more then it seems like we've wasted our time and may as well have used the RV32 version.

So I guess we should allow them and be clear that (e.g.) C is ignored if R=0 and W=0, and ASR is ignored if X=0
Which still allows future behaviour based on those combinations, we could say in future that they may "not be ignored".

How does that sound?

PRugg-Cap, 2024-05-02:
> The reason not to use the same encoding scheme for RV64 was to allow more combinations, so if we ban them all requiring a spin of the silicon to allow more then it seems like we've wasted our time and may as well have used the RV32 version.

Agreed, so I'd have assumed either (i) we ensure we're 100% certain the banned combinations are not useful (and make them properly reserved) or (ii) we treat all the combinations as equally valid (and allow ACPERM to create them). I don't understand why we'd have this middle ground where we prevent software accessing the bitpatterns conveniently, but allow it to access them with CBLD so we can't guarantee we can reuse those patterns without breaking some (admittedly esoteric) software?

tariqkurd-repo, 2024-05-08:
I'd vote for illegal combinations making the cap invalid, so CBLD would reject them.
So if those bit patterns do occur it will only be due to a future extension which is making use of them.

How does that sound?

PRugg-Cap, 2024-05-08:
Sounds good to me!

closed 2024-05-09 by tariqkurd-repo