riscv/riscv-cheri #254: Generic hardware revocation support?

jonwoodruff, 2024-05-17:
It is possible that CHERI-RISC-V implementations may have hardware acceleration for revocation, and it would be nice to have a generic extension that supports several strategies of hardware temporal safety/revocation acceleration.

For example, CHERIoT supports a hardware revocation bitmap such that, when memory is marked in the bitmap as "revoked", all capabilities that are loaded henceforth whose base points into the revoked range will have their tags cleared.  This has the property that the revoked memory becomes instantly impossible to reference.

In addition, one could imagine a DMA engine that sweeps memory, performing "capability revoked" checks concurrent to the main CPU.

In the way of an example, potentially, such an extension might allow incrementing the generation in a CSR, and indicating areas of revoked memory with a capability pointing to them.  The architecture might guarantee (either by hardware or m-mode software) that objects marked as revoked in generation - 2 is available to reuse. (Generation-1 would be "being swept/revoked now", and current generation is still being marked.)  An architecture similar to CHERIoT would need hardware support for painting the shadow map corresponding to the addresses in the capability presented in the new instruction.

Care might be taken to ensure that memory can be considered free to reuse without having to start another sweep, as the above design indicates.  Maybe a "all revocations done" CSR flag would do the trick, letting you know that Generation-1 is free to reuse.

davidchisnall, 2024-06-25:
For big cores, this absolutely does not want to be part of the core ISA. It is most likely to be implemented in the hardware data mover and so should not be specified until that has been specified architecturally (e.g. Intel QAT).

The CHERIoT design originated from scaling down a design for a 128-core system with CHERI, MTE, and a hardware data mover. The composition with MTE-like tagging is an important step because it means that all revocation state is architectural (expressed in the capability, memory colouring, or page-table state). Even then, there are a *lot* of subtle corner cases and there are some interesting implications on the memory model.

If you want to get to hardware revocation for MMU systems, start with memory colouring. Make sure you add the atomic decrement colour and check if zero instruction.

nwf, 2026-01-05:
> Is there any documentation to support this approach, and is it compatible with the current Svucrg extension?

See https://github.com/riscv/riscv-cheri/issues/340 .  The particular instruction to which David refers is detailed on slide 12 of the deck linked therein.

The SSM/MTE and MMU/PTE aspects of the design are almost orthogonal; so long as Svucrg works for _Cornucopia Reloaded_, it should work for the as-yet-hypothetical _Technicolor Cornucopia_.