riscv/riscv-cheri #263: ddc tag value unclear on debug mode entry

tomaird, 2024-05-23:
The spec says:

> Upon entry to debug mode, ddc is saved in dddc. ddc's metadata is set to the Infinite capability’s metadata and ddc's address remains unchanged.

Elsewhere in the spec, the tag bit is treated as a separate entity to the "metadata", so the above quote suggests that if `ddc` had its tag bit cleared before debug entry, that it would stay cleared after debug mode entry, even though the metadata is updated to the infinite cap.

Is this the intention? Or should it also have its tag set on debug mode entry when it gets the infinite cap metadata?

PRugg-Cap, 2024-05-23:
Agree this should be clarified: I don't think there's any point to changing its metadata if you don't also set the tag. The point is to make debug memory operations be easy to run without having to do a dance to find a capability (which you always can do anyway since `dinfc` exists).

andresag01, 2024-05-28:
Fixed via #267

closed 2024-05-28 by andresag01