riscv/riscv-cheri #273: Clarify effect of SCMODE and ACPERM on reserved encodings

PRugg-Cap, 2024-05-31:
(To avoid confusion, I'm using *unrepresentable* to mean permission combinations that don't have an encoding, e.g. any with `ASR` but no `X`, and *reserved* to mean an encoding that isn't constructible with `ACPERM`)

It's not defined what should happen if you try to change the M bit or permissions when the capability has a *reserved* permission encoding.

There's a note that I think addresses *unrepresentable* combinations: https://github.com/riscv/riscv-cheri/blob/f76f4a4f19297b4abc58bfaa62a087ab12542c13/src/insns/acperm_32bit.adoc?plain=1#L39-L40
(I interpret "are defined to return" to mean that you get the right result by following the legalisation rules, rather than that there's some other (implementation-defined?) definition somewhere)

`GCPERM` defines *reserved* encodings to return all zero permissions: https://github.com/riscv/riscv-cheri/blob/f76f4a4f19297b4abc58bfaa62a087ab12542c13/src/insns/gcperm_32bit.adoc?plain=1#L28-L29

It would be natural to add the same rule to the `ACPERM` decoding: https://github.com/riscv/riscv-cheri/blob/f76f4a4f19297b4abc58bfaa62a087ab12542c13/src/insns/acperm_32bit.adoc?plain=1#L31-L32

However, this has some surprising consequences, e.g. on RV64 if a capability's permissions field has reserved encoding `0b10000` (which would decode to `ASR=1, X=R=W=C=0`), trying to change the mode with `SCMODE` will cause the field to be re-encoded as `0b00000`. Since this capability can't be tagged, this probably isn't a huge issue.

We could also just insist that `SCMODE` is a nop if the permission encoding is *reserved*. This still means that `ACPERM` could potentially re-encode the field, e.g. in the above case, `ACPERM` with all 1s would still cause legalisation to `0b00000`, but I think that's less surprising.

Happy to file a PR if we agree on how to fix it, and sorry that there are so many annoying nits around this topic!

Timmmm, 2024-05-31:
I think it would be simultaneously worth clarifying what happens if you use `ACMODE` to remove X permission of a capability with M=1. Some examples would probably help a lot.

tariqkurd-repo, 2024-06-03:
@Timmmm I think you mean `ACPERM` - and the rule is specified: 

[image]

@PRugg-Cap I think the solution is probably that `SCMODE` is a `nop` if X-permission isn't set, as the only pre-requisite for setting M is that X=1.

Does that cover it?

PRugg-Cap, 2024-06-03:
@tariqkurd-repo almost. I think something like "if the permissions encoding is not reserved and the X bit is clear", though we've been writing "is not reserved" as "can be produced by `ACPERM`", which is fine. That then resolves the `SCMODE` issue.

There still needs to be a clarification on `ACPERM` though about how the decoding happens when the field is reserved. As a concrete example, if I `ACPERM` the illegal RV64 encoding `0b10111` (ASR without X) with `0b11111`, do I get `0b00000` (I refuse to decode the field, assume null perms, and reencode that, which is what I would assume) or `0b10111` (I continue to interpret the bits as usual, even in a reserved encoding)? Note that RV32 probably has to do the first one, because "continue to interpret the bits as usual" doesn't have an obvious meaning. In RV64, we'd be free to do the second one, but that may complicate repurposing those states later on.

tariqkurd-repo, 2024-06-03:
hmm interesting, I would go with the first one and return `0b0000` if `ACPERM` encounters reserved permissions.

PeterRugg, 2024-06-03:
Agreed: I think then we just need to add that as point `1. ii.` in the `ACPERM` rules. Something like "Reserved encodings give all null permissions".

To Tim's point, there's a slight oddity in the "common rules" that it has subpoints for each of the rules except M, telling you what to do, e.g. "1. ASR cannot be set without x being set i. Clear ASR unless X is set", but for point 3 it's just "3. M cannot be set without X being set" (though the subpoints seem kind of redundant anyway...)

tariqkurd-repo, 2024-06-03:
ok - PR updated - please have a look!

PRugg-Cap, 2024-06-03:
Yep, I think that resolves all my issues. It's slightly amusing that the `ACPERM` rules are now self-referential, but I think it's clear. Thanks!

tariqkurd-repo, 2024-06-03:
fixed in https://github.com/riscv/riscv-cheri/pull/271

closed 2024-06-03 by tariqkurd-repo